armbian/build #814: IGNORE_UPDATE Option will take compile fail.

nickfox-taterli, 2017-11-06:
at the first,i will compile with internet without IGNORE_UPDATE.
and does not modify anything,add IGNORE_UPDATE="yes" to config-default.conf,i will build fail.

[ o.k. ] Cleaning output/debs for [ nanopineo dev ]
[ o.k. ] Cleaning 
grep: /home/tater/x-armbian/build/cache/sources//Makefile: No such file or directory
grep: /home/tater/x-armbian/build/cache/sources//Makefile: No such file or directory
grep: /home/tater/x-armbian/build/cache/sources//Makefile: No such file or directory
grep: /home/tater/x-armbian/build/cache/sources//Makefile: No such file or directory

when IGNORE_UPDATE,i can't offline compile also.

igorpecovnik, 2017-11-06:
This IGNORE_UPDATE function is deprecated. Use CREATE_PATCHES instead. It will prompt you to modify sources and create a patch on that work. Use nanopineo NEXT branch.

zador-blood-stained, 2017-11-06:
IGNORE_UPDATES and CREATE_PATCHES are 2 completely different options, and IGNORE_UPDATES issue should be hopefully fixed after the last commit.

nickfox-taterli, 2017-11-06:
what's the different between IGNORE_UPDATES and CREATE_PATCHES ?

zador-blood-stained, 2017-11-07:
`IGNORE_UPDATES` is an option for internal use to skip downloading/checking sources and additional applications. It can work reliably only when used with a hidden option `USE_OVERLAYFS` and only after you downloaded the required sources.
`CREATE_PATCHES` is an option that allows creating a patch file from manual u-boot or kernel sources edits.

commit 2b0deaa7 (2017-11-06): Fix for IGNORE_UPDATES option not setting necessary variables
Closes #814
closed 2017-11-18 by zador-blood-stained