armbian/build #823: nanopi k2 does not support other hdmi res

nickfox-taterli, 2017-11-17:
I download image from https://www.armbian.com/nanopi-k2/ Ubuntu_xenial_default_desktop_nightly.7z

but it does not have hdmi output with 800x480,1024x600 or any other,it only support 1080p,720p,4k,i suggest to add hdmi res support,because 800x480,1024x600 hdmi screen are simple cheap.

my example code works well with 800x480:https://github.com/nickfox-taterli/amlogic-nanopi-k2/commit/60a64b80207d7547e1a2c43d1a80d37bc7d8e220

and HardKernel Odroid C2 offical image can works with many hdmi res.include 800x480,1024x600.

Tonymac32, 2017-11-21:
Hello nickfox-taterli,

Which Amlogic kernel did you start with?  For legacy Amlogic builds we use https://github.com/150balbes/Amlogic_s905-kernel/tree/armbian, which does not appear to be compatible with your patch.

nickfox-taterli, 2017-11-24:
i fork this https://github.com/friendlyarm/linux/tree/nanopi2-v4.4.y,make more output res support with https://github.com/nickfox-taterli/amlogic-nanopi-k2/commit/35e3091c1abd2bc123e4eda0e865356b35114c44 ,and i test https://github.com/150balbes/Amlogic_s905-kernel/tree/armbian also not support 800x480,1024x600 or any other res,only support 1080p,720p,4k also,i compare my patch and https://github.com/150balbes/Amlogic_s905-kernel/blob/armbian/drivers/amlogic/display/vout/tv_vout.h , i notice some code can be patch to support.

> static struct vmode_tvmode_tab_s mode_tab[]

> static struct vinfo_s tv_info[]

JackyYuenDacai, 2018-01-04:
is that because frame rate are not set? i also cant change resolution to others besides serveral mode with fixed frame rate , but via hdmi port.

Tonymac32, 2018-02-18:
NanoPi K2 S905 is being moved to Community Support, there won't be any official support moving forward other than what is coincidental with S905X support and community submitted.

closed 2018-02-18 by Tonymac32

Tonymac32, 2018-04-09:
@JackyYuenDacai @nickfox-taterli  Not reopening, but providing an update.  Meson64 image updates include patches from BayLibre that should provide additional screen resolutions. Armbian images have been updated,