armbian/build #830: kernel-headers: really call "make scripts" in postinst

smurfix, 2017-11-27:
The kernel-headers postinst says
```
cd /usr/src/linux-headers-4.14.2-mvebu64; echo "Compiling headers - please wait ..."; make -s scripts >/dev/null 2>&1
exit 0

```
… umm …
(a) `#!/bin/sh -e` is a good idea
(b) don't redirect error messages to /dev/null
(c) don't silently succeed when this step fails

One installs these headers in order to build out-of-tree modules. This is not possible without the binaries that get built here, thus installation should not succeed.

RossBencina, 2019-02-13:
I found @smurfix's comment (a) to be cryptic, so here is some more information:

The `-e` flag is the `errexit` flag. It will cause the script to exit if any statement returns a non-zero (error) return value. There is some future explanation [in this article entitled "Writing Robust Bash Shell Scripts"](https://www.davidpashley.com/articles/writing-robust-shell-scripts/), which says to use `set -e` at the top of the file rather than `#!/bin/sh -e`.

The [Debian Policy Manual ](https://www.debian.org/doc/debian-policy/ch-maintainerscripts.html) says of package maintainer scripts (preinst, postinst, prerm and postrm):
> The package management system looks at the exit status from these scripts. It is important that they exit with a non-zero status if there is an error, so that the package management system can stop its processing. For shell scripts this means that you almost always need to use set -e

lanefu, 2019-03-06:
closing issue see https://github.com/armbian/build/commit/bb2742ca98459c38f4a6859383ca5cc7116a0d99

RossBencina, 2019-03-06:
@lanefu why closing this issue? the problem remains with all targets, not just sunxi

lanefu, 2019-03-06:
@rossbencina

Sorry was cleaning up.  Reopening.

ioerror, 2020-01-15:
During some performance testing on various arm and arm64 machines, I've noticed that kernel headers are some broken. In hopes that it might assist others, I'm posting a follow up to this issue. I've had this issue with rockchip boards such as the tinkerboard and other more obscure systems.

My use case is straightforward. I need working kernel headers to build a cycle counter kernel module and then I load the module. On systems with broken headers, I am unable to build the module. One repo is for armv7 ( https://github.com/thoughtpolice/enable_arm_pmu ) and one is for armv8 ( https://github.com/rdolbeau/enable_arm_pmu ).

On some machines, the kernel headers work without intervention. On other machines, I am only able to build the module if I take the following steps:
```
cd /usr/src/linux-headers-`uname -r`/
export ARCH=`uname -m` # aka arm or arm64
make scripts
```
It is then usually possible to build `enable_arm_pmu` for the kernel:
```
cd enable_arm_pmu
make
./load-module
```

igorpecovnik, 2020-01-15:
> I've had this issue with rockchip boards such as the tinkerboard and other more obscure systems.

I assume on kernel 4.4.y ... and similar older kernels, but not from more recent kernels, derived from mainline?

mmriech, 2020-01-15:
> (a) `#!/bin/sh -e` is a good idea
> (b) don't redirect error messages to /dev/null
> (c) don't silently succeed when this step fails

Commit https://github.com/armbian/build/pull/1697/commits/f1937beeda38dcba9f6a5873450ec5cd11be5845 should have fixed (b) one month ago.

mmriech, 2020-01-19:
> (a) `#!/bin/sh -e` is a good idea
> (b) don't redirect error messages to /dev/null
> (c) don't silently succeed when this step fails
About c): This is a good idea, beacause at the moment the postinst fails silently if something is broken. It looks like other people already had the idea when they let the builddeb-script insert the line `set -e` into the postinst-script. But the final postinst-script does not contain the line anymore. Reason: The "general" kernel-patches of Armbian remove it again by calling `sed -e "s/set -e//g" -i $tmpdir/DEBIAN/postinst`. If we remove that command from the patches, the postinst wont't fail anymore silently. This would make a) obsolte.

smurfix, 2020-01-19:
> sed -e "s/set -e//g" 

WTF?

Fix errors instead of papering them over, dammit!

igorpecovnik, 2020-01-19:
> Fix errors instead of papering them over, dammit!

We are doing both. But different (private fork) kernel needs different approaches / patching. Not that simple as job from mainstream distributions ;)

mmriech, 2020-02-09:
> 
> 
> > Fix errors instead of papering them over, dammit!
> 
> We are doing both. But different (private fork) kernel needs different approaches / patching. Not that simple as job from mainstream distributions ;)

c) is fixed now after the merge of PR #1754 
b) is fixed as well (see https://github.com/armbian/build/issues/830#issuecomment-574816046)
a) What about this one? The postinst has the line `set -e` - isn't that the same like `#!/bin/sh -e`? If yes, I assume that this issue has finally been solved and can be closed now.

smurfix, 2020-02-10:
Closing. Thanks.

closed 2020-02-10 by smurfix