armbian/build #832: I cant see my screens left and bottom side...

beratuslu, 2017-11-29:
Hi,
I am using orange pi lite. and loaded ubuntu server..

my monitor is this: http://accessories.ap.dell.com/sna/productdetail.aspx?c=hk&l=en&s=bsd&cs=hkbsd1&sku=210-ADSZ

and I dont know how to fix. this Can someone help me?

Thank you.

igorpecovnik, 2017-11-29:
AFAIK there so no support for this kind of screen size in the kernel ... we added many but this: https://github.com/armbian/build/blob/master/patch/kernel/sun8i-default/z-0003-add-additional-video-modes.patch https://github.com/armbian/build/blob/master/packages/bsp/h3disp#L105-L236 try images with a modern kernel, currently under testing: https://dl.armbian.com/orangepilite/Debian_stretch_next_desktop_nightly.7z

beratuslu, 2017-11-29:
Hello Igor,

Thank for quick reply..
Do you have server version of it? I will use this as only server..
Cheers!

igorpecovnik, 2017-11-29:
No, not at the moment. Well, then you probably don't need a correct screen size :) ? Try that desktop image and if it works go to config and disable desktop. Remember, a kernel is still in testing stage and that is the reason for not providing more variants.

beratuslu, 2017-11-29:
ok thanks. I have loaded it up to how it looks and it looks good. But I have another problem. Which I will ask in another issue. Thank you.
Cheers!

closed 2017-11-29 by igorpecovnik