armbian/build #839: armbianmonitor - CPU temperature missing

pfeerick, 2017-12-05:
The ` [ ! -z "${SocTemp##*[0-9]*}" ]` test added in a recent commit https://github.com/armbian/build/commit/5cb346e80f2820569ead2bdf580b1f50706435c9#diff-9f87e1c4632903d6962052a358acebe1 to `armbianmonitor` has broken the temperature readings. Confirmed on at least three boards - OrangePiZero/legacy (me), pine64/legacy and odroidxu4/next (tkaiser). 

Basically, at the time the test is done at https://github.com/armbian/build/blob/master/packages/bsp/common/usr/bin/armbianmonitor#L346, the variable being used as a flag to indicate if the cputemp stuff should be done, and doesn't contain a valid temperature reading, so fails every time.  It's not until the subsequent line that the script attempts to load the temperature data, so if it was wanted, it would have to be placed further down and wrap around https://github.com/armbian/build/blob/master/packages/bsp/common/usr/bin/armbianmonitor#L348-L351, and there will be decimals involved at some point. The updatemotd code that was in that commit works fine though.

The corresponding test for the PMIC will probably break also.
https://github.com/armbian/build/blob/master/packages/bsp/common/usr/bin/armbianmonitor#L353

ThomasKaiser, 2017-12-05:
Obviously only tested with `/etc/update-motd.d/30-sysinfo` and not `armbianmonitor`. Due to missing comments I also fail to understand in which situations the code should do what so simply proposing to either start to communicate (add comments to code/commits so others understand later why stuff happened) or simply removing this stuff from `armbianmonitor` again. Preferably prior to 5.37 release.

igorpecovnik, 2017-12-05:
I am on the way to fix this.

closed 2017-12-05 by igorpecovnik

pfeerick, 2017-12-05:
Oh dear... I don't think Thomas will let you hear the end of this one for a while Igor! The fix is to remove the code which probably didn't get tested! At least the 🐛 has been 🔨 so now you can have 🍺  :-D :-D :-D

enzeperix, 2018-03-23:
Hello . I have noticed this problem again on ARMBIAN 5.38 stable Debian GNU/Linux 9 (stretch) 4.14.14-sunxi64 .
I use it on some Orange Pi Prime SBCs and on 80% of them for some reason the file /etc/armbianmonitor/datasources/soctemp  is missing . I tried to see what process is supposed to write in that file on a board which has the "soctemp" file but i couldn't find any . Sorry i'm new to Orange Pi and Armbian, if you need more info i can provide it :)

pfeerick, 2018-03-23:
IIRC there is a damon service that does some background data collection (think [this is the relevant bit](https://github.com/armbian/build/blob/master/packages/bsp/common/etc/init.d/armhwinfo)? ). `/etc/armbianmonitor/datasources/soctemp` is just a link though which points to the true soctemp value elsewhere in the virtual sys file system (just means the armbianmonitor script doens't have to deal will all the different possible locations of the info).