armbian/build #850: Option to exclude armbian-config from image?

akamensky, 2017-12-25:
Armbian-config is cool tool for someone who just began their adventure with SoC/SoM alt-arch based boards. However for someone more or less experienced with those (as well as with Linux systems) it is often useless (at least in my case as I prefer to do same things manually to make sure I can see all errors/warnings and be able to try out different options etc).

So I hope there would be an option to disable installation of this package into the image. Yes, I can manually change the list of packages installed to the image, but for unattended builds it is not practical.

igorpecovnik, 2017-12-25:
Use https://docs.armbian.com/Developer-Guide_User-Configurations/#user-provided-image-customization-script and `apt purge armbian-config` should do the job.

closed 2017-12-25 by igorpecovnik

akamensky, 2017-12-25:
@igorpecovnik Cool, that would work, but that first installs the package and then deletes it. Which is pure awesome. I would feel like my image is (\*ahem\*) haunted with ghosts of all packages first brought into this world and then mercilessly slaughtered in their cribs.