armbian/build #853: [espressobin] u-boot produces flash-image.bin that clocks CPU to 800 MHz instead of 1000 MHz

akamensky, 2017-12-26:
I've built u-boot previously from Marvell's repo and it produces correct image that clocks CPU to 1k MHz, the u-boot flash-image.bin (`cache/sources/arm-trusted-firmware-espressobin/atf-v1.3-armada-17.10/build/a3700/debug/flash-image.bin`) is clocked at 800 MHz instead.

Not a big deal as it still works and sees all the devices, but might be worth fixing.

If acceptable I may look into making a patch for this one.

igorpecovnik, 2017-12-26:
AFAIK some people have stability issues when clocking higher. I would say let's leave as is.

akamensky, 2017-12-26:
Might be different 3720 CPU version used on earlier boards (though I thought 3720 started at 1k lowest). If that's the case then I am okay with it, will use the one from Marvell's repo for my u-boot (that one clocks to 1k by default).

OK, to close as wontfix then, I guess.

igorpecovnik, 2017-12-26:
This issue is actually related to our build script - I choose safe to be a default one. We actually build all combinations, but manually. https://dl.armbian.com/espressobin/u-boot/ This has to be implemented to the build engine to run a cycle and produce them all. That part is already cluttered and I am not sure that I want to dig into that right now. Let's leave this open since it's not done as it should be.

akamensky, 2017-12-27:
Not sure if that is a good solution, but could possible create a sub-set of boards to build such as "espressobin-1000-1G" because some boards (not only espressobin) can come in different configurations - different CPU version (freq clock), different amount of RAM, etc.

That would make a menu list quite long tho, but that might be solved by making one more level of nesting, e.g. "root -> Orange Pi boards -> Orange Pi Plus 2" or "root -> Marvell boards -> Espressobin 1000 MHz 1G". But yes, not sure if that is better solution :)

zador-blood-stained, 2017-12-27:
Since currently u-boot images are not flashed on upgrade and we are relying on preinstalled u-boot I think building a single u-boot package with multiple binaries is a good option. Unfortunately the Espressobin requires a single u-boot blob with different ATF compile-time options (and ATF is built after the u-boot), and currently we only implemented the other scenario - different u-boot targets with a single ATF blob built before the u-boot.

igorpecovnik, 2018-09-23:
Now we make all flash images and pack them into u-boot package but flashing from Linux is not implemented yet. Issue from this topics is solved.

closed 2018-09-23 by igorpecovnik