armbian/build #863: [espressobin] GPIO_SYSFS not enabled by default?

akamensky, 2018-01-09:
I was trying to check GPIO pins on my ESPRESSObin and attach fan controlled with PWM, in kernel build if I go to check GPIO it appears already enabled, but I do not get anything GPIO related under `/sys`.

I suppose it is safe to assume GPIO is present on the board (since I never seen one without GPIO headers).

So would it be fine to enable it by default?

zador-blood-stained, 2018-01-13:
> So would it be fine to enable it by default?

Yes, should be fine to enable it by default on all 3 kernel branches. Feel free to send a PR.

> attach fan controlled with PWM

At least in the mainline sources I don't see anything that would resemble support for hardware PWM, so only external or software PWM will be available.

akamensky, 2018-01-14:
I will look into that then.

markusr, 2018-01-18:
The ```/sys/class/gpio/``` is deprecated. I think it makes sense to not enable it.

See: https://www.kernel.org/doc/Documentation/ABI/obsolete/sysfs-gpio
```
This ABI is deprecated and will be removed after 2020. It is
  replaced with the GPIO character device.
```

zador-blood-stained, 2018-01-18:
> The /sys/class/gpio/ is deprecated. I think it makes sense to not enable it.

It should still be enabled for now, and we may have LTS kernel branches where it will be always present. Unfortunately most userspace libraries, scripts and tutorials still use the sysfs API, and it can be tricky to use the gpiodev character device from a script.

akamensky, 2018-01-19:
@markusr I am with @zador-blood-stained on this. It is deprecated, does not mean it is not working or will disappear tomorrow (it is still as minimum 2 years until it is removed from kernel). And even after it is removed this new kernel should get adopted first.

akamensky, 2018-01-19:
I looked at simply enabling it, but seems there is something wrong with device tree definitions as it results in me seeing GPIO directories in SYSFS now, but it is not working as it should.

Any thoughts?

closed 2018-01-21 by akamensky