armbian/build #866: Wifi AP broken in WPA mode with Realtek 8188EUS - H3 legacy kernel

pushytoxin, 2018-01-18:
Board: custom board Orange Pi Plus
Using the legacy Linux 3.4.113 kernel and hostapd-realtek.
Hostapd access point works only with WPA disabled. When WPA is enabled, clients can not connect.

Building the module from [https://github.com/lwfinger/rtl8188eu](https://github.com/lwfinger/rtl8188eu) solves this issue.

This change can be included into Armbian by using a slightly modified version of the `packages/extras/rtl8192cu.sh` plugin script.

pushytoxin, 2018-01-18:
For the 8192cu plugin script to run, there was a condition that `$INSTALL_HEADERS` must be set. To me this seems unnecessary, the headers aren't needed during runtime.

pushytoxin, 2018-01-18:
#867

zador-blood-stained, 2018-01-20:
> To me this seems unnecessary, the headers aren't needed during runtime.

They are needed since you need to be building against the installed kernel headers, not against the kernel tree (since kernel packages may be cached during mass image building and sources in `$SRC/cache/sources/$LINUXSOURCEDIR/` may not match the kernel version installed to the image)

In any case this issue can be closed by #867

closed 2018-01-20 by zador-blood-stained