armbian/build #873: upstream broken

igorpecovnik, 2018-01-25:
4.14.15 breaks booting at least on sunxi devices, 4.14.14 works ... just found out.

mpmc, 2018-01-25:
Dang it! And I just started writing a 4.14.15 image >.<. Thanks for the heads up.

zador-blood-stained, 2018-01-25:
> 4.14.15 breaks booting at least on sunxi devices, 4.14.14 works ... just found out.

Logs? And what is affected (A20/H3/H5)? I don't see anything suspicious in the 4.14.15 changelog and won't be able to test this before the weekend.

igorpecovnik, 2018-01-25:
Yep, nothing suspicious in the changelog ... Problems might be older than 4.14.14 ... Well, I made two tests and it is not enough to conclude anything. Will get back on this later on.

igorpecovnik, 2018-01-28:
It looks it was some random issue, closing.

closed 2018-01-28 by igorpecovnik