armbian/build #876: compile.sh brings me 2 ERROR messages...

jiapei100, 2018-01-26:
Hi, @igorpecovnik 

I strictly followed your dooc from [https://docs.armbian.com/Developer-Guide_Build-Preparation/](https://docs.armbian.com/Developer-Guide_Build-Preparation/), however, I still got the following 2 ERROR messages:

```
[ error ] ERROR in function compile_uboot [ compilation.sh:108 ]
[ error ] Could not find required toolchain [ arm-linux-gnueabihf-gcc > 7.0 ]

```

I noticed that by just run **compile.sh** will automatically download a toolchain which is of version **5.4.0**, however, I've already installed my own toolchain from Linaro, which is of version **7.2.1**., and I'm quite sure this **7.2.1 toolchain** is under my **PATH**. 
So, where can I specify my own toolchain, instead of automatically downloading the toolchain from repository?

From my post from [https://github.com/friendlyarm/h3_lichee/issues/19](https://github.com/friendlyarm/h3_lichee/issues/19), the toolchain is automatically downloaded by default. Is there a single file, for instance, **system.sh** for us to specify our local toolchain and some other configurations???

Thank you very much
Pei

```
$ ./compile.sh 
[ o.k. ] Using config file [ config-default.conf ]
[ warn ] This script requires root privileges, trying to use sudo 
[sudo] password for jiapei: 
[ o.k. ] Using config file [ config-default.conf ]
[ o.k. ] This script will try to update

ssh: connect to host github.com port 22: Connection timed out
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ xenial ]
[ o.k. ] Updating from external repository [ aptly ]
[ o.k. ] Syncing clock [ host ]
/media/jiapei/Data/Downloads/operatingsystems/linuxdistro/armbian/lib/general.sh: line 579: ntpdate: command not found
[ .... ] Installing build dependencies 
Hit:1 http://packages.microsoft.com/repos/vscode stable InRelease
Hit:2 http://mirrors.aliyun.com/ubuntu xenial InRelease
Get:3 http://archive.ubuntukylin.com:10006/ubuntukylin xenial InRelease [18.1 kB]
Hit:4 http://mirrors.aliyun.com/ubuntu xenial-updates InRelease
Hit:5 http://mirrors.aliyun.com/ubuntu xenial-backports InRelease
Hit:6 http://mirrors.aliyun.com/ubuntu xenial-security InRelease
Hit:7 http://ppa.launchpad.net/lumas/photoqt/ubuntu xenial InRelease
Get:8 https://download.docker.com/linux/ubuntu xenial InRelease [65.8 kB]
Hit:9 http://repos.codelite.org/wx3.1.0/ubuntu xenial InRelease
Hit:10 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu xenial InRelease
Get:11 http://cran.stat.sfu.ca/bin/linux/ubuntu xenial/ InRelease [3,590 B]
Get:12 http://repo.aptly.info squeeze InRelease [4,867 B]
Get:13 http://repo.aptly.info squeeze/main amd64 Packages [3,153 B]
Get:14 http://repo.aptly.info squeeze/main i386 Packages [3,179 B]
Err:15 https://packagecloud.io/slacktechnologies/slack/debian jessie InRelease
  Connection timed out after 120944 milliseconds                                                                                                
Fetched 98.7 kB in 2min 4s (795 B/s)                                                                                                            
Reading package lists...                                                                                                                        
Building dependency tree...                                                                                                                     
Reading state information...                                                                                                                    
1 package can be upgraded. Run 'apt list --upgradable' to see it.
W: Failed to fetch https://packagecloud.io/slacktechnologies/slack/debian/dists/jessie/InRelease  Connection timed out after 120944 milliseconds
W: Some index files failed to download. They have been ignored, or old ones used instead.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  bbswitch-dkms containerd libatk-bridge2.0-dev libatspi2.0-dev
[... 204 lines cut ...]
Setting up libasan2-armhf-cross (5.4.0-6ubuntu1~16.04.4cross1) ...
Setting up libstdc++6-armhf-cross (5.4.0-6ubuntu1~16.04.4cross1) ...
Setting up libubsan0-armhf-cross (5.4.0-6ubuntu1~16.04.4cross1) ...
Setting up libgcc-5-dev-armhf-cross (5.4.0-6ubuntu1~16.04.4cross1) ...
Setting up gcc-5-arm-linux-gnueabihf (5.4.0-6ubuntu1~16.04.4cross1) ...
Setting up gcc-arm-linux-gnueabihf (4:5.3.1-1ubuntu1) ...
Setting up p7zip-full (9.20.1~dfsg.1-4.2) ...
Setting up pixz (1.0.6-2) ...
Setting up pv (1.6.0-1) ...
Setting up qemu-user-static (1:2.5+dfsg-5ubuntu10.16) ...
Setting up systemd-container (229-4ubuntu21) ...
Setting up u-boot-tools (2016.01+dfsg1-2ubuntu3) ...
Setting up aptly (1.2.0) ...
Setting up debootstrap (1.0.78+nmu1ubuntu1.5) ...
Setting up linux-libc-dev-armhf-cross (4.4.0-18.34cross1) ...
Setting up libc6-dev-armhf-cross (2.23-0ubuntu3cross1) ...
Setting up liblz4-tool (0.0~r131-2ubuntu2) ...
Processing triggers for systemd (229-4ubuntu21) ...
Processing triggers for ureadahead (0.100.0-19) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...
Processing triggers for initramfs-tools (0.122ubuntu8.10) ...
update-initramfs: Generating /boot/initrd.img-4.14.5-041405-generic
Processing triggers for dbus (1.10.6-1ubuntu3.3) ...
[ .... ] Downloading [ gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux ]
```

igorpecovnik, 2018-01-26:
> I've already installed my own toolchain from Linaro, which is of version 7.2.1., and I'm quite sure this 7.2.1 toolchain is under my PATH.

All you need to do is: get a blank Ubuntu server. Then run a script which takes care of everything. In case of network troubles, tampered Ubuntu installation or hacking the script at the very first try, you are on your own.

Script works:
https://pastebin.com/rjXjafwC

closed 2018-01-26 by igorpecovnik

jiapei100, 2018-01-26:
@igorpecovnik 
For simplicity, I just changed file **lib/general.sh** line **616** to

local toolchains=(
		"https://dl.armbian.com/_toolchains/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf.tar.xz"
		)

and, it still failed with the following **ERROR** message:

```
[ o.k. ] Building deb [ linux-u-boot-next-nanopineo_5.38_armhf.deb ]
[ error ] ERROR in function compile_uboot [ compilation.sh:232 ]
[ error ] Building u-boot package failed 
[ o.k. ] Process terminated 
```

It fails to build u-boot this time...

igorpecovnik, 2018-01-26:
https://pastebin.com/XUrBPCSB

Proved the second time - unattended build script works and this is the only possible case of support. If you want to build a kernel or image using this script - there is no need to fix it since it works perfectly fine. If you still want to fix it, go ahead, but don't expect help. We have enough problems without deliberately produced ones.