armbian/build #877: u-boot failed to build persisitently without any modification of armbian scripts.

jiapei100, 2018-01-26:
Hi, @igorpecovnik 

I persistently meet the following **ERROR** messages, without modifying any of the scripts.

```
[ o.k. ] Building deb [ linux-u-boot-next-nanopineo_5.38_armhf.deb ]
[ error ] ERROR in function compile_uboot [ compilation.sh:232 ]
[ error ] Building u-boot package failed 
[ o.k. ] Process terminated 

```

And result of 
`./compile.sh BOARD="nanopineo" BRANCH="next" KERNEL_ONLY="yes" KERNEL_CONFIGURE="no"` can be found at [https://pastebin.com/gBsr3Qdt](https://pastebin.com/gBsr3Qdt), while result of `./compile.sh` can be found at [https://pastebin.com/5DT4JmhD](https://pastebin.com/5DT4JmhD) ...

Please help...

Thank you 
Pei

zador-blood-stained, 2018-01-27:
> without modifying any of the scripts.

Your provided log says otherwise - you modified at least `lib/general.sh`, and you didn't provide any additional logs from `output/debug/` which may have additional info.

I have to repeat @igorpecovnik 's words - we can provide support only for **unmodified** build scripts located at https://github.com/armbian/build running in a build environment that is listed as supported. Issues with modified scripts, forks and with unsupported environments will be closed without a discussion and people who repeatedly ignore this may be blocked from opening issues in Armbian repositories.

jiapei100, 2018-01-27:
The **ONLY** modification by `git diff` is this part of **lib/general.sh**:

> $ git diff
> diff --git a/lib/general.sh b/lib/general.sh
> index a7fc07f..e33c394 100644
> --- a/lib/general.sh
> +++ b/lib/general.sh
> @@ -625,6 +625,7 @@ prepare_host()
>                 "https://dl.armbian.com/_toolchains/gcc-linaro-6.4.1-2017.11-x86_64_aarch64-linux-gnu.tar.xz"
>                 "https://dl.armbian.com/_toolchains/gcc-linaro-7.2.1-2017.11-x86_64_aarch64-linux-gnu.tar.xz"
>                 "https://dl.armbian.com/_toolchains/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf.tar.xz"
> +               #"/media/jiapei/Data/Downloads/programming/compilers/gcc/gcc-linaro/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf.tar.xz"
>                 )
>  
>         for toolchain in ${toolchains[@]}; do

And, I've got a lot log files under **output/debug** as follows:

> armbian/output/debug$ ls
> logs-26_01_2018-00_12_09.tgz  logs-26_01_2018-13_13_45.tgz  logs-26_01_2018-16_17_55.tgz  logs-26_01_2018-17_13_53.tgz  logs-26_01_2018-18_33_45.tgz  logs-26_01_2018-21_43_10.tgz  logs-.tgz   timestamp
> logs-26_01_2018-13_02_46.tgz  logs-26_01_2018-15_54_57.tgz  logs-26_01_2018-17_12_17.tgz  logs-26_01_2018-17_49_41.tgz  logs-26_01_2018-21_25_36.tgz  logs-26_01_2018-22_47_27.tgz  output.log
> 

which one do you want to see?

Cheers
Pei

Tonymac32, 2018-01-27:
Were you modifying the contents of the toolchains folder?  I just ran a nanopi neo NEXT build with no issue, no modifications of any files/folders/etc.

my toolchains folder:

`drwxr-xr-x 8 root root 4096 Dec  6 20:14 gcc-linaro-4.9.4-2017.01-x86_64_aarch64-linux-gnu
drwxr-xr-x 8 root root 4096 Dec  6 20:14 gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabi
drwxr-xr-x 8 root root 4096 Dec  6 20:15 gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf
drwxr-xr-x 8 root root 4096 Dec  6 20:16 gcc-linaro-5.5.0-2017.10-x86_64_aarch64-linux-gnu
drwxr-xr-x 8 root root 4096 Dec  6 20:17 gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabi
drwxr-xr-x 8 root root 4096 Dec  6 20:18 gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf
drwxr-xr-x 8 root root 4096 Dec  6 20:20 gcc-linaro-6.4.1-2017.11-x86_64_aarch64-linux-gnu
drwxr-xr-x 8 root root 4096 Dec  6 20:19 gcc-linaro-6.4.1-2017.11-x86_64_arm-linux-gnueabihf
drwxr-xr-x 8 root root 4096 Dec  6 20:21 gcc-linaro-7.2.1-2017.11-x86_64_aarch64-linux-gnu
drwxr-xr-x 8 root root 4096 Dec  6 20:22 gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf
drwxr-xr-x 7 root root 4096 Dec  6 20:13 gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux
`

zador-blood-stained, 2018-01-27:
> which one do you want to see?

output.log, it should contain output from failed packaging commands

jiapei100, 2018-01-27:
@zador-blood-stained 

Thank yo uso much for your prompt reply... The following is my cached toolchain folder.

```
armbian/cache/toolchains$ ls -ls
total 3880
3852 -rwxrwxrwx 1 jiapei jiapei 3944448 Jan 27 01:29 gcc-linaro-4.9.4-2017.01-x86_64_aarch64-linux-gnu.tar.xz
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 26 23:11 gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 26 23:28 gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabi
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 27 00:15 gcc-linaro-6.4.1-2017.11-x86_64_aarch64-linux-gnu
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 26 23:58 gcc-linaro-6.4.1-2017.11-x86_64_arm-linux-gnueabihf
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 27 00:50 gcc-linaro-7.2.1-2017.11-x86_64_aarch64-linux-gnu
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 26 18:51 gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf
   4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 26 21:51 gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux
```

What's more, I did notice that my host Ubuntu **automatically installed** a toolchain of version **5.4.0** .

Any further suggestions?

Thank you ....
Pei

jiapei100, 2018-01-27:
@zador-blood-stained 

Thank you ...

My output.log content is:

```
armbian/output/debug$ cat output.log
Displaying message: Command line: setting BOARD to nanopineo info
Displaying message: Command line: setting BRANCH to next info
Displaying message: Command line: setting KERNEL_ONLY to yes info
Displaying message: Command line: setting KERNEL_CONFIGURE to no info
Displaying message: Preparing host info
Displaying message: Build host OS release xenial info
Displaying message: Syncing clock host info
Displaying message: Downloading gcc-linaro-4.9.4-2017.01-x86_64_aarch64-linux-gnu
```

Cheers
Pei

zador-blood-stained, 2018-01-27:
> My output.log content is:

It should contain more info, looks like you aborted your last build before the first toolchain was downloaded. Please restart the compilation, wait until it exits with an error (if it does) and check output.log then.

zador-blood-stained, 2018-01-27:
>    `4 drwxrwxrwx 1 jiapei jiapei    4096 Jan 26 23:11 gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabihf`

For some reason your toolchain directories have non-standard permissions (777) though on stock Ubuntu those permissions should be 755 or 775. Such permissions on the build script directory will cause `dpkg-deb` to fail.

jiapei100, 2018-01-28:
@zador-blood-stained 

```
armbian/output/debug$ cat output.log 
Displaying message: Command line: setting BOARD to nanopineo info
Displaying message: Command line: setting BRANCH to next info
Displaying message: Command line: setting KERNEL_ONLY to yes info
Displaying message: Command line: setting KERNEL_CONFIGURE to no info
Displaying message: Preparing host info
Displaying message: Build host OS release xenial info
Displaying message: Syncing clock host info
Displaying message: Downloading gcc-linaro-4.9.4-2017.01-x86_64_aarch64-linux-gnu
Displaying message: Verifying
Displaying message: Verification failed  wrn
Displaying message: Downloading gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabi
Displaying message: Verifying
Displaying message: Verification failed  wrn
Displaying message: Downloading gcc-linaro-5.5.0-2017.10-x86_64_aarch64-linux-gnu
Displaying message: Verifying
Displaying message: Extracting
Displaying message: Download complete  info
Displaying message: Downloading gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf
Displaying message: Verifying
Displaying message: Verification failed  wrn

## BUILD SCRIPT ENVIRONMENT

Repository: 
Version: f8709ec

Host OS: xenial
Host arch: amd64
Host system: Linux jiapei-GT72-6QE 4.14.5-041405-generic #201712101332 SMP Sun Dec 10 13:33:10 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

## Build script directories
Build directory permissions:
# file: /media/jiapei/Data/Downloads/operatingsystems/linuxdistro/armbian
# owner: jiapei
# group: jiapei
user::rwx
group::rwx
other::rwx

Temp directory premissions:
# file: /media/jiapei/Data/Downloads/operatingsystems/linuxdistro/armbian/.tmp
# owner: jiapei
# group: jiapei
user::rwx
group::rwx
other::rwx

## BUILD CONFIGURATION

Build target:
Board: nanopineo
Branch: next
Desktop: no

Kernel configuration:
Repository: git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
Branch: branch:linux-4.14.y
Config file: linux-sunxi-next

U-boot configuration:
Repository: git://git.denx.de/u-boot.git
Branch: tag:v2017.11
Config file: nanopi_neo_defconfig

Partitioning configuration:
Root partition type: ext4
[... 39 lines cut ...]
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] add-sunvell-r69.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] add-tritium.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] add_emmc_orangepiwin.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] adjust-default-dram-clockspeeds.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] adjust-small-boards-cpufreq.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] enable-DT-overlays-support.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] enable-autoboot-keyed.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] fdt-setprop-fix-unaligned-access.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] fix-sunxi-gpio-driver.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] fix-usb1-vbus-opiwin.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] h3-Fix-PLL1-setup-to-never-use-dividers.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] h3-enable-power-led.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] h3-set-safe-axi_apb-clock-dividers.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] lower-default-DRAM-freq-A64-H5.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] lower-default-cpufreq-H5.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] sun8i-set-machid.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] video-fix-vsync-polarity-bits.patch  info
Displaying message: * [\e[32ml\e[0m][\e[33mb\e[0m] workaround-reboot-is-poweroff-olimex-a20.patch  info
Displaying message: Building deb linux-u-boot-next-nanopineo_5.38_armhf.deb info
dpkg-deb: error: control directory has bad permissions 777 (must be >=0755 and <=0775)
Displaying message: ERROR in function compile_uboot compilation.sh:232 err
Displaying message: Building u-boot package failed  err
Displaying message: Process terminated  info

```

zador-blood-stained, 2018-01-28:
> dpkg-deb: error: control directory has bad permissions 777 (must be >=0755 and <=0775)

As I suspected, build directory and its subdirectories has non-standard permissions, even though with the default umask they should be 755 or 775. Please fix the permissions manually.

jiapei100, 2018-01-28:
@zador-blood-stained 
Hey, Zador... Thank you .... 
This is more or less a kind of weird to me that 777 won't work ?? **FIRST** time happened to me...

Well, let's move on. After I modify the permissions to 755, which command should I go ? I can see from my side that if I do **./compile.sh** again, it starts downloading the toolchain once again, and will also change the permission back to 777 again?

So, is there any command for me to **continue from a point from when I do NOT need to download the toolchains once and once again**??

```
armbian$ ./compile.sh BOARD="nanopineo" BRANCH="next" KERNEL_ONLY="yes" KERNEL_CONFIGURE="no"
[ o.k. ] Using config file [ config-default.conf ]
[ warn ] This script requires root privileges, trying to use sudo 
[sudo] password for jiapei: 
[ o.k. ] Using config file [ config-default.conf ]
[ o.k. ] This script will try to update
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
[ o.k. ] Command line: setting BOARD to [ nanopineo ]
[ o.k. ] Command line: setting BRANCH to [ next ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ yes ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ no ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ xenial ]
[ o.k. ] Syncing clock [ host ]
[ .... ] Downloading [ gcc-linaro-4.9.4-2017.01-x86_64_aarch64-linux-gnu ]
```

Thank you very much
Pei

zador-blood-stained, 2018-01-28:
The build script should work without such issues on a freshly installed Ubuntu with the standard Ubuntu kernel and when running on a native Linux filesystem (ext4, btrfs, xfs, ...). I cannot guess which modifications you did to the kernel or rootfs or what filesystem are you using since you are not using the build script from armbian/build.
There are no chmod calls in the script that touch the "other" permission bits and the build script sets umask 002.
Even if there are ways to prevent some of the issues I will not try to fix them since they can be avoided by using a clean Ubuntu installation or virtualization methods like Virtualbox.

jiapei100, 2018-01-28:
@zador-blood-stained 
Did you mean the host Ubuntu and host kernel? Well, that's right... I'm NOT using the default host kernel. I'm using the mainline kernel at [http://kernel.ubuntu.com/~kernel-ppa/mainline/](http://kernel.ubuntu.com/~kernel-ppa/mainline/) .

What's more, even if the toolchain is of permission 777, will it influence the installation that much? Due to my naive understanding, 755<775<777, which should **ALL** work, and in most cases it's like that, because 777 has even **higher** permission...

And, **NOW**, it still fails with exactly the same **ERROR** message:
```
[ o.k. ] Building deb [ linux-u-boot-next-nanopineo_5.38_armhf.deb ]
[ error ] ERROR in function compile_uboot [ compilation.sh:232 ]
[ error ] Building u-boot package failed 
[ o.k. ] Process terminated 
```

zador-blood-stained, 2018-01-28:
> What's more, even if the toolchain is of permission 777, will it influence the installation that much?

`dpkg-deb` expects specific safe permissions for its control files, and packaging system files with such permissions (world-writable) would also be unsafe. It is not our tool so we cannot control or change its behavior but at least I agree that we should not try to bypass this check.

> dpkg-deb: error: control directory has bad permissions 777 (must be >=0755 and <=0775)

jiapei100, 2018-01-28:
@zador-blood-stained 
Thank you Zador...
So, if I understand right, I will have **ONLY** one way out **NOW**:
change back to the kernel from repository, instead of using the **mainline kernel** ?

Let me try
Pei

zador-blood-stained, 2018-01-28:
I'm not sure if your issues come from the kernel (which is unlikely) or anything else like external FS and its mounting options, but I can say for sure that the build script should work fine on a fresh Ubuntu installation with no OS customizations.

jiapei100, 2018-01-30:
Well, I guess I found the problem, Probably it's because all files/folders on the mounted external hard drive are automatically set to permission 777.

So, my solution: git clone to my primary hard drive and I noticed all files under **cache/toolchains** are **NOT** of permission 777.

BTW, I think it is possible that if I install **Ubuntu Xenial** from https://www.armbian.com/nanopi-neo/, I should be able to build the **NEWEST** u-boot and kernel from within NanoPi NEO, right? even if it's quite slow ?? Any support on this?

Cheers
Pei

zador-blood-stained, 2018-01-30:
> I should be able to build the NEWEST u-boot and kernel from within NanoPi NEO, right?

Only manually (clone sources, patch them and compile), this script is supported only on x86_64 hardware due to specific dependencies.

jiapei100, 2018-02-01:
@zador-blood-stained 
Finally, done... 4.14.16

My god....

Thank you very much zador...
Pei

closed 2018-02-01 by jiapei100