armbian/build #879: Vagrant provision script uses wrong user

akamensky, 2018-02-01:
Official Vagrant Ubuntu 16.04 no longer uses `ubuntu` user. (v20171011.0.0, v20180126.0.0 and all newer ones I presume)

That was non-standard and they changed it to comply with vagrant guidelines (which requires default user to be `vagrant`).

Thus it gets a little confusing when on newer images provisioning script clones to /home/ubuntu/armbian, but when ssh into the box the user is vagrant. Took me solid 10 minutes to figure out what's happening :)

akamensky, 2018-02-01:
Also JFYI, new Ubuntu 16.04 images come without python2 (python3 is new default).

If there is no dependency on python2, I suggest fixing username and setting a version constraint (to use newer images `config.vm.box_version >= 20171011.0.0`

zador-blood-stained, 2018-02-02:
> If there is no dependency on python2

There is a dependency that should be resolved automatically (hopefully).

closed 2018-02-02 by zador-blood-stained

akamensky, 2018-02-02:
To be fair I built a couple of images (espressobin and orangepi plus 2) and it worked fine. So should be good for the rest I think

commit 00599c91 (2018-02-02): Set box version constraint to make sure using latest one and change username to default "vagrant" user (#880)

Fixes #879