armbian/build #891: Prepare SD card using compiled deb files

Almaz1c, 2018-02-15:
Hello!
May be it is obvious for all peoples but I can't understand it.
I got `linux-u-boot-next-orangepizeroplus2-h5_5.40_arm64.deb` and other deb files after compiling from sources using `.\compile.sh`. How to prepare bootable SD card to plug it in Orangepi zero board?
I expected to get `.bin`, `.cpio.gz` and so on files.

igorpecovnik, 2018-02-15:
If you want Debian .bin/.iso image you need to build an image, not just kernel and u-boot packages. Those debs are prepared to be deployed standard way on (our) system. If you want to install them manually, you need to extract .deb file and proceed with instructions(post-install script) found inside the deb file. Those tips might be helpful: https://docs.armbian.com/User-Guide_Advanced-Features/#how-to-unbrick-the-system

closed 2018-02-15 by Almaz1c

Almaz1c, 2018-02-15:
Thank you for explanation!