armbian/build #895: Tinker Board Error compiling WiFi drivers in default kernel (4.4)

gleam2003, 2018-02-17:
Error on build:

[ o.k. ] Installing additional driver [ RT8188EU ]
[ o.k. ] Checking git sources [ rt8188eu master ]
[ .... ] Up to date 
1 out of 1 hunk FAILED -- saving rejects to file include/ieee80211.h.rej
cp: impossibile creare il file regolare '/home/gleam/build/.tmp/rootfs-default-tinkerboard-jessie-no/lib/modules/4.4.115-rockchip/kernel/net/wireless/': Non è una directory
[ o.k. ] Installing additional driver [ RT8192 ]
[ o.k. ] Checking git sources [ rt8192cu master ]
[ .... ] Skipping checkout 
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file include/ieee80211.h.rej
make[2]: *** [/home/gleam/build/cache/sources/rt8192cu/os_dep/linux/os_intfs.o] Errore 1
make[1]: *** [_module_/home/gleam/build/cache/sources/rt8192cu] Errore 2
make: *** [modules] Errore 2
cp: impossibile eseguire stat di '*.ko': File o directory non esistente

Wifi after build dosen't work

igorpecovnik, 2018-02-17:
Those drivers are optional and they do not work on all kernels. We tend to move such 3rd party wireless drivers inside kernel - if this is possible without severe changes to the driver code, if someone needs and prepare this, if we are bored, ... It is an added value over added value :) 

But image should boot even those errors pops out.

nintendontit, 2018-02-17:
ok about additional drivers but onboard wifi dosent work, ifconfig report no wireless devices.

Tonymac32, 2018-02-18:
I will retest this, however as I commented in the commit, making changes to the kernel config around the "rkwifi" subsystem is tricky and typically results in build errors due to.  I'm running a 5.41 tinker build right now with onboard wifi.  Double check to make sure you haven't accidentally changed any of the selections around the rkwifi system.

nintendontit, 2018-02-18:
About RT8188 and mt7601 failed not to compile but copy compiled file to kernel because distination directory:
lib/modules/4.4.115-rockchip/kernel/net/wireless/
dosen't exist, I change it to:
lib/modules/4.4.115-rockchip/kernel/drivers/net/wireless/

nintendontit, 2018-02-18:
rtl8192cu compilation failed

igorpecovnik, 2018-02-18:
> rtl8192cu compilation failed

Those drivers are optional and we don't try to fix them. They work on some kernels and that's why they are here. If you need one for your board, [find a better driver out there, at wireless drivers specialists, and try to compile on board](https://docs.armbian.com/User-Guide_Advanced-Features/#how-to-build-a-wireless-driver). This issue tracker is for problems with the built environment, not the driver code itself. It is way out of our power and project aims.

nintendontit, 2018-02-19:
Yes Sorry but I think change directory for the other 2 driver's is small fix that can be added. Write about rtl8192cu only for specific that only the other 2 works not for ask help.

Tonymac32, 2018-02-19:
Verified default kernel has functional onboard wifi.

nintendontit, 2018-02-19:
Yes confirm after update and flush all files, thank you

closed 2018-02-21 by gleam2003