armbian/build #910: Ramdisk image is corrupt or invalid

nathanshearer, 2018-03-16:
It looks like all the images being created do not work (or are corrupt) for the Orange Pi PC2:

Debian server - mainline kernel (Testing)
 - "Ramdisk image is corrupt or invalid"
 - does not boot

https://dl.armbian.com/orangepipc2/Ubuntu_xenial_next.7z
 - "Ramdisk image is corrupt or invalid"
 - does not boot

https://dl.armbian.com/orangepipc2/nightly/Armbian_5.34.171121_Orangepipc2_Ubuntu_xenial_next_4.13.14.7z
 - "Ramdisk image is corrupt or invalid"
 - does not boot

https://dl.armbian.com/orangepipc2/nightly/Armbian_5.37.171217_Orangepipc2_Debian_stretch_next_4.14.6.7z
 - "Ramdisk image is corrupt or invalid"
 - does not boot

Screenshots of the error are below:
 - https://imgur.com/N7Fkm9u
 - https://imgur.com/JL8aOzl

igorpecovnik, 2018-03-16:
Nothing wrong here, check your hardware, sdcard, powering, ...
https://docs.armbian.com/User-Guide_Getting-Started/

Most of the images are tested and I test this one again, build log of Ubuntu Xenial next: http://ix.io/XTG

closed 2018-03-16 by igorpecovnik

nathanshearer, 2018-03-16:
You are right, my usb power supply, usb cable, and sd card were all okay, but the orange pi itself was the problem. I changed it out for my other orange pi pc2 board and it worked perfectly.

Thanks for the help