armbian/build #915: Olimex Lime2 emmc board freezes within a day

Timple, 2018-03-30:
We have an issue where the Olimex Lime2 emmc board freezes within a day. The nand versions seems to work fine for already a week now.

Usually we compile our own images using the build scripts creating ubuntu server images for the Olimex Lime2 nand and emmc versions. When we noticed the freezing of emmc boards we used the standard Armbian_5.38_Lime2_Debian_stretch_next_4.14.14 images from the downloads page which has the same result.

In the past (last year) we did not have reliability issues with the emmc boards, it might be related to this commit?: https://github.com/armbian/build/commit/d50a55e970102d1e616676bf8b8a170172889e47

Debugging is hard because the network is down, the serial interface is down and even the blinking cursor on the screen stops blinking. Looking into the logs of the SD card afterwards did not reveal anything.

Are there any options we can try to extend debugging? Our could you try to run an image on the emmc board for a couple of days?

Regards, Tim

igorpecovnik, 2018-04-02:
Merging configuration should not affect this problem. Board freezing is in most cases related to DRAM clocked too high - this can be adjusted in u-boot but people report that going back to the older kernel, helps. I am not sure if they changed u-boot as well. Check this thread: https://forum.armbian.com/topic/6386-lime2-mainline-kernel-with-debian-9-stretch-becomes-unresponsive-forced-reboot-required/

Timple, 2018-12-23:
Not experiencing this anymore with new images

closed 2018-12-23 by Timple