armbian/build #916: Sub-5Mbps on Gige with 4.16.0-rc3 on RK3328 / Rock64

itdaniher, 2018-03-31:
USB3 performs fine. Ethernet is crippled. Ethtool doesn't indicate an issue.

```
root@rock64:~# uname -a
Linux rock64 4.16.0-rc3-rk3328 #6 SMP Wed Mar 28 20:05:25 CEST 2018 aarch64 GNU/Linux
```

```
------------------------------------------------------------
Client connecting to 192.168.2.129, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.2.125 port 39764 connected with 192.168.2.129 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.1 sec  5.38 MBytes  4.45 Mbits/sec
```

```
root@rock64:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: ug
        Wake-on: d
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: yes
```

itdaniher, 2018-04-02:
Bad Rock64!

Tried a second Rock64, 914Mbit. Not your issue!

itdaniher, 2018-04-02:
Issue isn't so simple. I saw slow (15Mbps) speeds again, although a reboot cleared it on this device whereas it persisted through multiple reboots and cables on the last rock.

There seems to be some issue still with the beta kernel's ethernet performance as my original issue indicated.

jebivetar, 2018-04-04:
i have a similar issue open in ayufan's repo - i believe armbian does use ayufan's kernel.

i thought it was a bad rock at first too, and mine seems to have other issue (e.g., it's practically bricked at this point in time), but really the ethernet speed issue is widlyl intermittent

Tonymac32, 2018-04-04:
Yes, Armbian uses the Ayufan kernel.  I know that a lot of work was put into the ethernet tx/rx delays for the 4.4 kernel, I would assume those were carried over to the 4.16-rc but you would need to check the device tree to be sure, looking quickly they match.  

Can you give the armbianmonitor -u link?  I will build a new image for mine as soon as I'm able and see what happens.

jebivetar, 2018-04-04:
Am not of much more help - my rock64 is a brick at the moment, hopefully it's fixable.

-------- Original Message --------

itdaniher, 2018-04-07:
https://github.com/ayufan-rock64/linux-mainline-kernel/commit/73e42e18669934fa96cf2bb54291da54177076d7

Looks possibly related?

Tonymac32, 2018-04-07:
> ayufan-rock64/linux-mainline-kernel@73e42e1
>
> Looks possibly related?

Most likely.  GbE has been touchy on that hardware.

itdaniher, 2018-06-28:
4.17.0-rc6-rk3328

closed 2018-06-28 by itdaniher