armbian/build #918: motd storage polling method

jebivetar, 2018-04-04:
a little unsure if this is the right place to report this but the distinctions between the various venues (forum, here, etc) are a little nebulous

anyhow

currently the MOTD script seems to poll /dev/sda for its "storage" infochunk, which very often isn't the primary storage on armbian devices - it's usually some mmc device on a sdcard - and /dev/sda is some attached drive.

ergo, this method is

a) inaccurate - for me it gives me the free space of a random device, since I have multiple drives attached via USB and fstab mounts them differently every time. it could perhaps be better to poll all the devices and then tally it up, although i personally wouldn't like that
b) inefficient - motd generation, and therefore login, is very slow since it waits for the drive to needlessly spin-up to provide useless information

igorpecovnik, 2018-04-04:
Thanks for those ideas. I agree that this was not done properly and is considered some leftover from early builds which were, at that stage, made for personal usage. But nevertheless, it's configurable what you want to see: https://github.com/armbian/build/blob/master/packages/bsp/common/etc/update-motd.d/30-sysinfo#L17-L18 Perhaps we only need some better configuration method?

ThomasKaiser, 2018-04-04:
> inefficient - motd generation, and therefore login, is very slow since it waits for the drive to needlessly spin-up to provide useless information

Neither `df -h` nor `hddtemp` should wake up a sleeping HDD. I personally never liked the motd approach at all (since delaying the login) but I fail to see this problem. Can you elaborate on what happens exactly? Using a second terminal sending the disk to sleep with `hdparm` and then executing both `df -h /dev/sda1` and `hddtemp -u C -nq /dev/sda1` and report which one of those wakes up your disk?

ThomasKaiser, 2018-04-12:
No feedback given. Closing.

jebivetar, 2018-05-01:
why would you close a ticket after one week? no one's being paid to respond to tickets 24/7

1. log in via ssh
2. blank terminal
3. wait 3-5s while sda1 spins up
4. motd appears

df -h does not wake up individual disks; hddtemp does, however motd has never displayed hdd temperatures for me regardless of usb->sata device

igorpecovnik, 2018-05-04:
It works as expected. Drive is sleeping ... SSH login ... it displays this way, while there is no info about the temp if drive is sleeping. 

``` _____                                   _     _       
| ____|___ _ __  _ __ ___  ___ ___  ___ | |__ (_)_ __  
|  _| / __| '_ \| '__/ _ \/ __/ __|/ _ \| '_ \| | '_ \ 
| |___\__ \ |_) | | |  __/\__ \__ \ (_) | |_) | | | | |
|_____|___/ .__/|_|  \___||___/___/\___/|_.__/|_|_| |_|
          |_|                                          

Welcome to ARMBIAN 5.44.180504 testing Ubuntu 18.04 LTS 4.4.131-mvebu64   
System load:   0.24 0.27 0.17   Up time:       6 min
Memory usage:  7 % of 926MB     IP:            172.16.100.189
Usage of /:    7% of 15G        storage/:      90% of 2.7T   

[ 0 security updates available, 1 updates total: apt upgrade ]
Last check: 2018-05-04 06:18

Last login: Fri May  4 06:24:34 2018 from 172.16.100.175

root@espressobin:~# hddtemp /dev/sda
/dev/sda: ST3000DM001-1CH166: drive is sleeping

ThomasKaiser, 2018-05-04:
@igorpecovnik obviously your disk is already in `/etc/hddtemp.db` while @jebivetar's isn't. For `hddtemp` to be of any help a string line in it's 'database' is needed. Since this package hasn't been updated since almost a decade most modern HDD and SSD are missing of course. Same with USB disks where the enclosure is not SAT compliant. You can't query such a disk with hddtemp anyway since not up to the job

To get a disk 'supported' `hddtemp /dev/sda --debug` has to be executed and if it works (SATA or USB/SAT) then this exact string including whitespace has been appended to `/etc/hddtemp.db` in a compatible way.

So IMO either fix this or remove the whole functionality since of no value anyway (if people are concerned about their disk's temperature they need real monitoring running all the time or 'highest peak since last login' and not just current temperature when logging in. The latter is almost fooling inexperienced users)

ThomasKaiser, 2018-05-04:
Voting for removing the misleading functionality entirely.

igorpecovnik, 2018-05-04:
> Voting for removing the misleading functionality entirely.

Removing HDD temperature display? Just do it and remove package as well.

ThomasKaiser, 2018-05-04:
Only changed in development branch. I have no idea based on what commits get merged to master (and why stuff e.g. originating from Olimex still directly goes into master)

closed 2018-05-04 by ThomasKaiser

igorpecovnik, 2018-05-04:
> Only changed in development branch. I have no idea based on what commits get merged to master (and why stuff e.g. originating from Olimex still directly goes into master)

OK. Well, it's still a bit messy with this devel/master change. Well, if someone sends PR mistakenly to the master and if its irrelevant change, I accept PR rather than requesting changes. But sooner or later some major merge should be executed ... I am not ready yet.

commit f73485b9 (2018-05-04): Remove HDD temp display from motd
Closes #918