armbian/build #919: Unable to compile NanoPi K2

nickfox-taterli, 2018-04-06:
u-boot-nanopik2-s905.bin not found when compile uboot,

when i change file name from nanopik2-s905.csc to nanopik2.csc,it will be fix.

igorpecovnik, 2018-04-06:
Have you used development branch of the build script?

nickfox-taterli, 2018-04-06:
I used master branch for this.but when i compile with development branch,it will automatic switch to master too.

igorpecovnik, 2018-04-06:
You need to add LIB_TAG="development" to the config-default.conf 

In case you still are not able to do so, do a clean git clone.

nickfox-taterli, 2018-04-08:
same error when LIB_TAG="development"

root@tater-VirtualBox:~/build# cat config-default.conf 
# Read build script documentation http://www.armbian.com/using-armbian-tools/
# for detailed explanation of these options and for additional options not listed here

LIB_TAG="development"

  CC      lib/tables_csum.o
  CC      lib/string.o
  CC      lib/time.o
  CC      lib/uuid.o
  CC      lib/rand.o
  CC      lib/vsprintf.o
  CC      lib/panic.o
  CC      lib/strto.o
  CC      lib/strmhz.o
  LD      lib/built-in.o
  CC      examples/standalone/stubs.o
  CC      examples/standalone/hello_world.o
  LD      examples/standalone/libstubs.o
  LD      examples/standalone/hello_world
  OBJCOPY examples/standalone/hello_world.bin
  OBJCOPY examples/standalone/hello_world.srec
  LD      u-boot
  OBJCOPY u-boot-nodtb.bin
  DTC     arch/arm/dts/meson-gxbb-odroidc2.dtb
start=$(ccache aarch64-linux-gnu-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(ccache aarch64-linux-gnu-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x01000000 $start $end
make[2]: 'arch/arm/dts/meson-gxbb-odroidc2.dtb' is up to date.
  SHIPPED dts/dt.dtb
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  MKIMAGE u-boot-dtb.img
[ error ] ERROR in function compile_uboot [ compilation.sh:168 ]
[ error ] U-boot file not found [ u-boot-nanopik2-s905.bin ]
[ o.k. ] Process terminated 
root@tater-VirtualBox:~/build#

igorpecovnik, 2018-04-08:
>In case you still are not able to do so, do a clean git clone.

[ error ] U-boot file not found [ u-boot-nanopik2-s905.bin ]

This file was renamed from
packages\blobs\meson\u-boot-nanopik2.bin
to
packages\blobs\meson\u-boot-nanopik2-s905.bin

If you have made changes to build script sources they don't update ...

nickfox-taterli, 2018-04-09:
that is ok.

closed 2018-04-09 by nickfox-taterli