armbian/build #925: boot failed (tty log inside) [ Full OS image -> do not change the kernel config -> nanopineo -> default -> xenial ]

Sfinx, 2018-04-11:
```
U-Boot SPL 2017.11-armbian (Apr 11 2018 - 10:11:38)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2017.11-armbian (Apr 11 2018 - 10:11:38 +0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi NEO
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3708 bytes read in 196 ms (17.6 KiB/s)
## Executing script at 43100000
U-boot loaded from SD
Boot script loaded from mmc
142 bytes read in 149 ms (0 Bytes/s)
5219954 bytes read in 656 ms (7.6 MiB/s)
4752384 bytes read in 633 ms (7.2 MiB/s)
Found legacy kernel configuration
34416 bytes read in 441 ms (76.2 KiB/s)
## Loading init Ramdisk from Legacy Image at 43300000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    5219890 Bytes = 5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK

Starting kernel ...

[sun8i_fixup]: From boot, get meminfo:
        Start:  0x40000000
        Size:   512MB
ion_carveout reserve: 160m@0 256m@0 130m@1 200m@1
ion_reserve_select: ion chipid  [0x2c00081!
ion_reserve_common: ion reserve: [0x56000000, 0x60000000]!
<Loading, please wait...
starting version 229
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
done.
Begin: Will now check root file system ... fsck from util-linux 2.27.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p1] fsck.ext4 -a -C0 /dev/mmcblk0p1 
/dev/mmcblk0p1 has unsupported feature(s): metadata_csum
e2fsck: Get a newer version of e2fsck!
fsck exited with status code 8
done.
Warning: File system check failed but did not detect errors
mount: mounting /dev/mmcblk0p1 on /root failed: Invalid argument
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... mount: mounting /dev on /root/dev failed: No such file or directy
done.
mount: mounting /run on /root/run failed: No such file or directory
run-init: current directory on the same filesystem as the root: error 0
Target filesystem doesn't have requested /sbin/init.
run-init: current directory on the same filesystem as the root: error 0
run-init: current directory on the same filesystem as the root: error 0
run-init: current directory on the same filesystem as the root: error 0
run-init: current directory on the same filesystem as the root: error 0
run-init: current directory on the same filesystem as the root: error 0
No init found. Try passing init= bootarg.
Rebooting automatically due to panic= boot argument
[   19.106191] Restarting system.
[   19.109671] Restarting Linux version 3.4.113-sun8i (root@Yasha) (gcc version 5.5.0 (Linaro GCC 5.5-208
[   19.109693] 
```

The SD card was verified after writing, so mounted it shows the init present:

```
root@Yasha:/mnt# file sbin/init 
sbin/init: symbolic link to /lib/systemd/systemd
root@Yasha:/mnt# file lib/systemd/systemd
lib/systemd/systemd: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=8554a7237b90b238a2d3e3ab825cb3cfb008d7cc, stripped
root@Yasha:/mnt# 
```

igorpecovnik, 2018-04-11:
Two things are possible. Deboostrap or your SD card failed. Are you using master or development branch?

Sfinx, 2018-04-11:
As I've already told - the SD card was verified after write. git status:

```
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
```

git log:

```
commit 032c789aa146f1f8e009bf2f067a15eceb2bd0c3
Merge: 8a69e66 9efdeaa
Author: Igor Peд█ovnik <igorpecovnik@users.noreply.github.com>
Date:   Wed Apr 11 10:10:51 2018 +0200

    Merge pull request #921 from daijh223/master
    
    Add support for Firefly-ROC-RK3328-CC
```

What branch should use people to have the working image ?

igorpecovnik, 2018-04-11:
We need to investigate ... Builds logs would help. They are in output/debug

Sfinx, 2018-04-11:
Build logs attached

[logs-11_04_2018-10_09_50.tar.gz](https://github.com/armbian/build/files/1898530/logs-11_04_2018-10_09_50.tar.gz)

igorpecovnik, 2018-04-11:
Nothing suspicious in logs. Will try to reproduce this problem by building an image ... you can try deleting cache/rootfs  then add LIB_TAG="development" to config-default.conf and repeat the process. I assume you are building on some not very altered Ubuntu Xenial x64?

Sfinx, 2018-04-11:
Ok, trying now with deleted cache. First I've builded the 4.x + debian for the same board, but there no SPI working so I've tried to build the 3.x kernel but it is appeared that it even do not boots. Hoping this is cache/rootfs issue...

igorpecovnik, 2018-04-11:
I recreate image and it builds and boot fine. With stock settings.
https://pastebin.com/1JAyC1Cx

Try removing cache/roofs only

Sfinx, 2018-04-11:
Still building... BTW: seeing a bunch of :

```
qemu: Unsupported syscall: 384
qemu: Unsupported syscall: 384
```

This is stock x86_64 Ubuntu 16.04 LTS

igorpecovnik, 2018-04-11:
> Still building... BTW: seeing a bunch of :

https://github.com/armbian/build/issues/924
Fixed in a development build.

zador-blood-stained, 2018-04-11:
> `Host system: Linux Yasha 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

Host OS does not appear to be a clean Ubuntu Xenial installation. I have the exact kernel version on a stock Ubuntu 17.10, however container environment was not detected.

> ` /dev/mmcblk0p1 has unsupported feature(s): metadata_csum`

This should not happen unless something is messed up like an unsupported version of e2fsprogs shipped by newer Ubuntu/Debian releases.

Sfinx, 2018-04-11:
BTW: your pastebin shows that you compiled the image for nanopiair but the issue is about nanopineo.

Sfinx, 2018-04-11:
Deleting cache/rootfs do not helps - exactly the same failed boot log. Now trying to re-build from all-cleaned state...

Sfinx, 2018-04-11:
> > `Host system: Linux Yasha 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`
>
> Host OS does not appear to be a clean Ubuntu Xenial installation. I have the exact kernel version on a stock Ubuntu 17.10, however container environment was not detected.

Users can select the kernel at Ubuntu, I've selected 4.13 - this is totally unrelated to the build/boot fail problem.

...
# cat /etc/*release

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"
NAME="Ubuntu"
VERSION="16.04.4 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.4 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
....

>
> > ` /dev/mmcblk0p1 has unsupported feature(s): metadata_csum`
>
> This should not happen unless something is messed up like an unsupported version of e2fsprogs shipped by newer Ubuntu/Debian releases.

# e2fsck -V
e2fsck 1.43.3 (04-Sep-2016)
	Using EXT2FS Library version 1.43.3, 04-Sep-2016

zador-blood-stained, 2018-04-11:
> Users can select the kernel at Ubuntu, I've selected 4.13 - this is totally unrelated to the build/boot fail problem.

Yes, you can install anything on anything if you want. The question is whether the resulting environment was tested (no) and can issues in it be supported (no)

> e2fsck -V
> e2fsck 1.43.3 (04-Sep-2016)
>	Using EXT2FS Library version 1.43.3, 04-Sep-2016

https://packages.ubuntu.com/xenial/e2fsprogs - 1.42.x
https://packages.ubuntu.com/artful/e2fsprogs - 1.43.x

So, as I said, not a clean Xenial installation.

zador-blood-stained, 2018-04-11:
The problem lies in default FS options in `/etc/mke2fs.conf`. At least `64bit` and `metadata_csum` options do not work well with u-boot and old kernels, and in Xenial (the only supported host OS currently) these options are not set by default.

Sfinx, 2018-04-11:
I will try to build with vagrant or so.

Sfinx, 2018-04-12:
image builded with vagrant booted ok, so closing the issue as invalid

closed 2018-04-12 by Sfinx