armbian/build #936: No kernel configuration when target file exists in output/debs

ryanlederman, 2018-04-19:
**Summary:** I had previously built a kernel by choosing the 'U-boot and kernel packages' option, then chose my board, kernel and selected to show and edit the configuration before building.

**Expected result:** The `Kernel Configuration` screen appears to make changes to the config.

**Actual result:** The build script finishes quietly without any opportunity to change the config. Here is the output after choosing `Show a kernel configuration menu before compilation`:

> [ o.k. ] Downloading sources 
> [ o.k. ] Checking git sources [ u-boot v2017.11 ]
> [ .... ] Up to date 
> [ o.k. ] Checking git sources [ linux-sun8i sun8i ]
> [ .... ] Up to date 
> [ o.k. ] Checking git sources [ sunxi-tools master ]
> [ .... ] Up to date 
> [ o.k. ] Checking git sources [ rkbin-tools master ]
> [ .... ] Up to date 
> [ o.k. ] Checking git sources [ marvell-tools A3700_utils-armada-17.10 ]
> [ .... ] Up to date 
> [ o.k. ] Checking git sources [ odroidc2-blobs master ]
> [ .... ] Up to date 
> [ o.k. ] Kernel build done [ @host ]
> **[ o.k. ] Target directory [ /home/ryan/build/output/debs/ ]
> [ o.k. ] File name [ linux-image-sun8i_5.41_armhf.deb ]**
> [ o.k. ] Runtime [ 0 min ]

On a hunch, I did the following, which remedied it for me:

`rm -rf output/debs && ./compile.sh`

Please let me know if this is a bug; if it is, I will try to repro here and provide a patch.

igorpecovnik, 2018-04-19:
Your deb files are cached. Add "debs" to CLEAN_LEVEL="make,oldcache" in your config-default.conf and it will be rebuilt.

ryanlederman, 2018-04-19:
@igorpecovnik I see; forgive me if I don't understand the process properly, but it seems like if you are trying to reconfigure a kernel deb, it should at least warn you that you've got it cached and it won't be rebuilt. Is there a particular reason one would not want it to be rebuilt when choosing to change the config?