armbian/build #944: net.ifnames=0 for u-boot

norrest, 2018-04-29:
Hi
sometimes i have problem - dont work ethernet, because adapter name was changed
So how change/add "net.ifnames=0" 
to u-bood or build script?

norrest, 2018-04-29:
**net.ifnames=0 biosdevname=0**

norrest, 2018-05-02:
```
sed -i 's/disp.screen0_output_mode=1920x1080p60/monitor=none net.ifnames=0 biosdevname=0/' /boot/boot.cmd
mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr 
```
Should help!

igorpecovnik, 2018-05-04:
Try by adding extraargs=net.ifnames=0 biosdevname=0 to /boot/armbianEnv.txt

closed 2018-05-04 by igorpecovnik

norrest, 2018-05-04:
It turns out that editing 
`/boot/boot.scr`
not properly?

norrest, 2018-05-05:
I cant find file /boot/armbianEnv.txt
I need create it?

norrest, 2018-05-28:
Up!

igorpecovnik, 2018-05-28:
This is build script issue tracker and there is nothing wrong with it. Support for such things: https://forum.armbian.com
You must have run some very old image that armbianEnv is not present and implemented. It's also not done yet on all kernels.

norrest, 2018-05-28:
At the moment the problem is not in the boot script!
I can not understand what to do with the absence of the armbianEnv.txt file
I need create an empty file and write there ( extraargs = net.ifnames = 0 biosdevname = 0) ?

igorpecovnik, 2018-05-28:
The problem you have is not related to the armbian build script. Go to the forum and explain the problem there. Yes, you can create the file but you also need to have boot script that is recent enough.