armbian/build #945: Orange Pi One: ERR:set cpu frequency to 1296MHz failed!

AlexanderPoleschuk, 2018-04-30:
I've installed https://dl.armbian.com/orangepione/Ubuntu_xenial_default_desktop.7z build on Orange Pi One yesterday.
As soon as I start kodi 18 alpha build (which is not a root cause, I believe, but produces a high load to the system).
I have kern.log and syslog both slamed with only one type of message:
Apr 30 09:13:10 localhost kernel: [44018.037389] [cpu_freq] ERR:set cpu frequency to 1296MHz failed!
Apr 30 09:13:10 localhost kernel: [44018.237217] [ARISC ERROR] :message process error
Apr 30 09:13:10 localhost kernel: [44018.237245] [ARISC ERROR] :message addr   : f004b840
Apr 30 09:13:10 localhost kernel: [44018.237266] [ARISC ERROR] :message state  : 5
Apr 30 09:13:10 localhost kernel: [44018.237284] [ARISC ERROR] :message attr   : 2
Apr 30 09:13:10 localhost kernel: [44018.237304] [ARISC ERROR] :message type   : 30
Apr 30 09:13:10 localhost kernel: [44018.237323] [ARISC ERROR] :message result : ff
Apr 30 09:13:10 localhost kernel: [44018.237341] [ARISC WARING] :callback not install

This results in log2ram folder to be occupied very fast, so 50M of 50M are eaten mostly by kernel and syslogs...

zador-blood-stained, 2018-04-30:
Output of `armbianmonitor -u` please. This frequency is not available on Orange Pi One and this limit is set at least in 2 places, so something must have changed the system configuration.

AlexanderPoleschuk, 2018-04-30:
http://ix.io/1981

Not sure if you find it useful though... I've already ran the solution described here:
https://michlstechblog.info/blog/orangepi-high-cpu-temperature-or-arisc-errors-in-kernel-log/

zador-blood-stained, 2018-04-30:
Your log is from an Orange Pi PC image with script.bin relinked to an Orange Pi One version. Of course this will cause issues.

> I've installed https://dl.armbian.com/orangepione/Ubuntu_xenial_default_desktop.7z build on Orange Pi One yesterday.

I just downloaded, unpacked and mounted this image. Your log is from a different image.

AlexanderPoleschuk, 2018-04-30:
yes, my bad! sorry about that...

closed 2018-04-30 by AlexanderPoleschuk