armbian/build #947: Feature request: Full root encryption via LUKS (cryptsetup)

zciendor, 2018-04-30:
When using Armbian for development boards only, the lack of root encryption isn't probably a big deal. However, when using Armbian on more "consumer friendly" / "production ready" boards like the `Odroid HC1/HC2` for live/production use cases, **root-encryption is a MUST**.

### Problem: 
Adding a LUKS container **after** the Armbian image has been built or even flashed is almost impossible. 

### Solution:
The best way is to adjust the Armbian build script to prepare a flashable image with a 2-partition layout:
- small, unencrypted boot partition containing `cryptsetup` in the initramfs to unlock the root partition
- LUKS container on the second partition for the root file system

### Challenges:
1. Boards like the Odroid HC1/HC2 don't even have a display connector. How do you povide the passphrase to unlock the root filesystem during boot? One solution is to include SSH (dropbear) in the initramfs, so users can provide the passphrase from a remote system using SSH before the actual Armbian system ist booted.
2. Since the boot system is unencrypted, the dropbear ssh server must use a different host key then the encrypted (secure) root system. However, having different host keys for the same hostname/ip address will cause "host key changed" warnings on the SSH client computer and one must constantly modify `~/.ssh/known_hosts`. This can be solved by running dropbear (boot) and OpenSSH (root) on different ports. From a security (obscurity) perspective, it's slightly better to run OpenSSH on a different port than the default one (22). This protects against lazy port scans.
3. The resulting image contains the `LUKS volume key` and the `passphrase` to unlock that key. If the resulting image is re-distributed to other users than the person who built the image, those users still have to generate a new volume key and re-encrypt the root partition before flashing.
4. Resizing the root filesystem on first run to use the full capacity of the SD card it was flashed on is slightly more effort. The LUKS container must be resized first, before the root filesystem can be expanded.

zciendor, 2018-04-30:
### Initial implementation
Development branch: https://github.com/zhiverbox/armbian-build/tree/development
Upstream pull request: https://github.com/armbian/build/pull/948

The above pull request solves all these challenges by introducing 4 additional and **optional** build options:
- `CRYPTROOT_ENABLE=yes`
- `CRYPTROOT_PASSPHRASE="MYSECRECTPASS"`
- `CRYPTROOT_SSH_UNLOCK=yes`
- `CRYPTROOT_SSH_PORT=2222`

The implementation has been tested on the `Odroid HC1 (XU4)` with `Debian stretch` using `ext4` and `btrfs`. Automatic resizing (resize2fs service) has been modified accordingly and works for both `ext4` and `btrfs`.

Looking forward for testing, feedback and hopefully upstream acceptance

rosbeef, 2018-05-13:
I aprove 
I'm actualy doing it manualy on a armbian stretch 4.9 linux kernel:
- flash an SD
- run the odroid
- install lvm2 cryptsetup dropbear-initramfs. 
- share my laptop public  key to dropbear-initramfs authorised_key configfile
- create a crypted ssd then a lvm2 part on it 
- rsync the sd rootfs partition
- modify the crypttab, fstab and boot.ini
- reboot 
That process could be run with a script like nand-sata-install

I would to run that process with a 4.14 linux kernel but it doesn't work and i dont know why. 
i don't plan to by an uart to usb so if you can help me.

thanks for all your jobs

closed 2018-08-31 by igorpecovnik