armbian/build #964: Rock64 stops booting after upgrade?

itdaniher, 2018-05-17:
uboot says:

```
U-Boot 2017.09-armbian (May 12 2018 - 13:36:13 +0200)

Model: Pine64 Rock64
DRAM:  4 GiB
MMC:   rksdmmc@ff520000: 0, rksdmmc@ff500000: 1
*** Warning - bad CRC, using default environment

In:    serial@ff130000
Out:   serial@ff130000
Err:   serial@ff130000
Model: Pine64 Rock64
misc_init_r
cpuid=55524b50303930303800000000080f0a
serial=1ed13b519e8e90f
Net:   eth0: ethernet@ff540000
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
mmc_init: -95, time 10
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
2923 bytes read in 28 ms (101.6 KiB/s)
## Executing script at 00500000
Boot script loaded from mmc 1
239 bytes read in 22 ms (9.8 KiB/s)
## Error: Can't overwrite "ethaddr"
himport_r: can't insert "ethaddr=02:38:a0:82:2d:71" into hash table
## Error: Can't overwrite "eth1addr"
himport_r: can't insert "eth1addr=02:38:a0:82:2e:c1" into hash table
6070964 bytes read in 785 ms (7.4 MiB/s)
19224584 bytes read in 2385 ms (7.7 MiB/s)
56906 bytes read in 122 ms (455.1 KiB/s)
** File not found /boot/dtb/rockchip/overlay/rk3328-fixup.scr **
## Loading init Ramdisk from Legacy Image at 04000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    6070900 Bytes = 5.8 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Ramdisk to fc936000, end fcf00274 ... OK
   reserving fdt memory region: addr=1f00000 size=74000
   Loading Device Tree to 00000000fc8bf000, end 00000000fc935fff ... OK

Starting kernel ...

ERROR:   rockchip_plat_sip_handler: unhandled SMC (0x82000003)
```

then it hangs there for a prolonged period of time

itdaniher, 2018-05-18:
after "starting kernel" eventually prints a bit more...

```
Starting kernel ...

ERROR:   rockchip_plat_sip_handler: unhandled SMC (0x82000003)
DDR version 1.06 20170424
In
SRX
LPDDR3
786MHz
Bus Width=32 Col=11 Bank=8 Row=15/15 CS=2 Die Bus-Width=32 Size=4096MB
ddrconfig:7
OUT
Boot1 Release Time: 2017-05-18, version: 2.43
ChipType = 0x11, 187

```

itdaniher, 2018-05-18:
dropped in a /boot from the latest stable, it took after patching armbianEnv with the right uuid.

did a full apt-get upgrade and it went south again....

itdaniher, 2018-05-18:
aka, Armbian_5.42_Rock64_Ubuntu_xenial_default_4.4.124.7z works, but I'm not so sure whatever I'm pulling from apt is sane.

itdaniher, 2018-05-18:
did have beta.armbian.com configured, moving back to apt.armbian.com I suppose

igorpecovnik, 2018-05-18:
> did have beta.armbian.com configured, moving back to apt.armbian.com I suppose

Beta is in an unknown state and since you brought this up will be inspected. A few days ago I was playing with Rock64 and everything was normal.

closed 2018-05-18 by igorpecovnik