armbian/build #968: Customized /etc/network/interfaces overwritten on first boot

ryanlederman, 2018-05-18:
As requested:

The config that I wrote to `/etc/network/interfaces` during execution of `userpatches/customize-image.sh` was overwritten by the following at `packages/bsp/common/etc/init.d/firstrun:121` (_there is similar code at line 114 too_):

```bash
cd /etc/network/ && ln -sf interfaces.network-manager interfaces
```
- This only runs for certain board types: `"Orange Pi Lite"|"Orange Pi Zero Plus 2"|"NanoPi Air"|"Orange Pi R1" and "Orange Pi Zero"|"NanoPi Duo"|"Sunvell R69"`.
- It would be nice to have an option to exclude network-manager from the image completely, or at least disabled by default.

**Workaround**
Executing the following in `userpatches/customize-image.sh` will remove those commands from the firstrun script:

```bash
sed -i -e 's/ln\(.*\)interfaces/echo "NOOP"/gm' /etc/init.d/firstrun
```

ryanlederman, 2018-05-22:
**Addendum**
I am also trying to customize `/etc/fstab` in order to persist NFS mount points, but `lib/debootstrap-ng.sh:386` does:

```bash
rm -f $SDCARD/etc/fstab
```
Which obviously will undo the lines I wrote. Suggest to agree on a comment like `#!noerase` above a line that should stay in `/etc/fstab` after this script runs instead of deleting it entirely.

ryanlederman, 2018-05-23:
> That should not be a problem since customization comes later in the process. Debootstrap only prepares initial rootfs.

My mistake. You are correct.

ryanlederman, 2018-05-23:
> That should not be a problem since customization comes later in the process

@igorpecovnik forgive me if I don't understand how this works, but let me explain the behavior I am seeing and I'll leave it up to you.

As far as I can tell, `customize_image` is only called once anywhere in in a script under `lib/`. I'm including the surrounding code as well from `debootstrap-ng.sh` starting at line `67` (comments added by me):

```bash
customize_image # <-- sources userpatches/customize-image.sh

# clean up / prepare for making the image
umount_chroot "$SDCARD"
post_debootstrap_tweaks

if [[ $ROOTFS_TYPE == fel ]]; then
	FEL_ROOTFS=$SDCARD/
	display_alert "Starting FEL boot" "$BOARD" "info"
	source $SRC/lib/fel-load.sh
else
	prepare_partitions # <-- rm -f $SDCARD/etc/fstab
	create_image
fi
```
In my code, in `customize-image.sh`, I write my NFS mount to `/etc/fstab`. In order to be sure, I added a `cat` to the code that deletes `$SDCARD/etc/fstab` . As you can see here, the lines I wrote during the customization are in *that file*:

```
[ o.k. ] Creating partitions [ root: ext4 ]
# UNCONFIGURED FSTAB FOR BASE SYSTEM
/dev/mmcblk0p1 / ext4 defaults 0 1
/dev/mmcblk0p2 /usr ext4 defaults 0 2
#nfs
/usr/local/src/sparkage	/export/sparkage	none	bind	0	0
```
So, for now, I am doing my workaround as detailed previously. If I don't do that, my NFS mount is not present in the image when the build process is completed.

igorpecovnik, 2018-05-24:
Aha, I see. You are right. We can't just move that function call below ...  you can do this mount by adding things to https://github.com/armbian/build/blob/development/packages/bsp/common/etc/init.d/firstrun

This way NFS mount will be added at first run.

closed 2018-07-13 by igorpecovnik

ryanlederman, 2018-09-07:
@igorpecovnik that file doesn't seem to exist in the repo anymore. What's the current appropriate way to do this?

igorpecovnik, 2018-09-07:
Located elsewhere: https://github.com/armbian/build/tree/master/packages/bsp/common/usr/lib/armbian

ryanlederman, 2018-09-07:
Thanks! Also, I've moved my `userpatches/customize-host.sh` into a freshly checked-out directory, and I noticed that `userpatches` doesn't exist in the current commit.

I still see a message that says `Calling image customization script` but it doesn't seem to run?