armbian/build #983: Latest debian stretch server image is unbootable on TinkerBoard

Ntemis, 2018-05-29:
Compiled a fresh image twice with same outcome

Tonymac32, 2018-05-29:
Do you have UART console output to see where it goes wrong?

Ntemis, 2018-05-30:
LOL found out whats the issue
Even though a user selects tinkerboard as target he always get 
# DO NOT EDIT THIS FILE
#
# Please edit /boot/armbianEnv.txt to set supported parameters
#

setenv rootdev "/dev/mmcblk0p1"
setenv fdt_file "rk3288-miqi.dtb"
setenv ramdisk_addr_r "0x21000000"
setenv console "ttyS2,115200n8"
setenv verbosity "1"

itest.b ${devnum} == 0 && echo "U-boot loaded from SD"
itest.b ${devnum} == 1 && echo "U-boot loaded from eMMC"

if load ${devtype} ${devnum}:1 ${ramdisk_addr_r} /boot/armbianEnv.txt || load ${devtype} ${devnum}:1 ${ramdisk_addr_r} armbianEnv.txt; then
	env import -t ${ramdisk_addr_r} ${filesize}
fi

setenv bootargs "consoleblank=0 scandelay root=${rootdev} rw console=${console} rootfstype=ext4 loglevel=${verbosity} rootwait usb-storage.quirks=${usbstoragequirks} ${extraargs}"
ext4load ${devtype} ${devnum}:1 ${fdt_addr_r} /boot/dtb/${fdt_file} || fatload ${devtype} ${devnum}:1 ${fdt_addr_r} dtb/${fdt_file} || ext4load ${devtype} ${devnum}:1 ${fdt_addr_r} dtb/${fdt_file}
ext4load ${devtype} ${devnum}:1 ${ramdisk_addr_r} /boot/uInitrd || fatload ${devtype} ${devnum}:1 ${ramdisk_addr_r} uInitrd || ext4load ${devtype} ${devnum}:1 ${ramdisk_addr_r} uInitrd
ext4load ${devtype} ${devnum}:1 ${kernel_addr_r} /boot/zImage || fatload ${devtype} ${devnum}:1 ${kernel_addr_r} zImage || ext4load ${devtype} ${devnum}:1 ${kernel_addr_r} zImage
bootz ${kernel_addr_r} ${ramdisk_addr_r} ${fdt_addr_r}
# mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr

Ntemis, 2018-05-30:
Will provide uart output tomorrow

Tonymac32, 2018-05-30:
well, right, the bootscript loads armbianEnv.txt which overrides the miqi dtb selection.  Unless something else changed it's been that way for a year now

Tonymac32, 2018-05-30:
Just built and booted 4.4.133, Armbian 5.46, booted with no issues.

Ntemis, 2018-05-30:
Yes i found out that this is the commit that leaves me unbootable :(
While it shouldn't  i"ll change android version to 0x08000000 and try again.
DO NOT MERGE atm

dmschwartzii, 2018-05-31:
@Tonymac32 anyway I can get a download of what you built?

igorpecovnik, 2018-05-31:
@Ntemis 

https://forum.armbian.com/topic/7346-cant-boot-tinker-board-s-from-emmc/?do=findComment&comment=55356

Ntemis, 2018-05-31:
I booted up @igorpecovnik  :)

igorpecovnik, 2018-05-31:
With midgard for Android?

Ntemis, 2018-05-31:
Yeap uploading atm for the group to start building TinkerPie

Ntemis, 2018-05-31:
Used http://dpaste.com/1VPK8VX with https://github.com/armbian/build/pull/981

Ntemis, 2018-05-31:
I think will close now

closed 2018-05-31 by Ntemis

Tonymac32, 2018-05-31:
Does the touchscreen still work with the change to the way the Mipi_dsi?

Ntemis, 2018-05-31:
I dont have any to tell but you can test
https://androidfilehost.com/?fid=890278863836288820

chwe17, 2018-05-31:
Does JMCC know about this issue here? @ntemis did you test how much this affects his '[RK3288 MEDIA TESTING SCRIPT](https://forum.armbian.com/topic/7262-rk3288-media-testing-script/)'? You have the board, you can test it... ;)
IMO it would be really annoying when we break his whole work with switching to 'android_mali' only cause 'linux_mali' is currently broken. For me, his 'UN-official, UN-supported, UN-necessary, UN-popular, UN-precedented...' multimedia script multimedia script is of more importance than making 'default Armbian' suitable for 'retropie images' (as you're suggest with https://github.com/armbian/build/pull/981). In case both is possible, perfect. But in case this means that his whole script gets obsolete and he needs to start from plain, then I favor that we wait/look for a fix of the linux mali driver.

Tonymac32, 2018-05-31:
Well, the issue here is actually Rockchip updating the DDK and somehow breaking the compile.  The switch was then made to the Linux driver, but it is an older version yet, so it doesn't function for users.  I want to get rockchip stable and see if I either have to clone their repo or pick a "moment in time via a tag/commit number so we don't get surprises.  As of today I would assume the media script is not working either with a new build.

https://github.com/rockchip-linux/kernel/commit/91842c9d8b4f21a431643d97269d142d9524cd9a#diff-60ad033154c5f79f2acf5a2d6576198e

dmschwartzii, 2018-05-31:
@all in my testing using midgard for android breaks emulationstation which kind of defeats the purpose

chwe17, 2018-05-31:
> As of today I would assume the media script is not working either with a new build.

not my main argument.. ;) 

> then I favor that we wait/look for a fix of the linux mali driver.

he spend a huge amount of time, bringing this features to armbian, not (yet) officially but with support and fixing the script for currently available armbian images. If this work gets obsolete just by:
```
+CONFIG_MALI_MIDGARD_FOR_ANDROID=y
+# CONFIG_MALI_MIDGARD_FOR_LINUX is not set
```
and:
> that gcc-wrapper script also adds to the kernel compile time, so pretty nice to get rid of :)

than it's IMO not worth to break his work fully for future armbian images (I know it's currently broken) only to make 'new builds' suitable for RetroPi. Armbians buildscript allows them to build their own kernel/images with minor efforts and to avoid breaking their builds they can freeze the kernel by default and/or use` customize-image.sh` for whatever they want to do with a 'RetroPi image based on Armbian' similar as @ThomasKaiser does it for the OMV images he/OMV provides.

igorpecovnik, 2018-05-31:
> it's IMO not worth to break

Ofc not. Possible choices:

- go back in time as Tony proposed and we have a working build again
- wait that Rockhip engineers fix what they recently broke
- do nothing :)

chwe17, 2018-05-31:
> IMO not worth to break his work **fully for future armbian images**

I don't know how much 'android mali' differs from 'linux mali'. But if this has noticeable side-effects (means it breaks stuff like JMCCs multimedia work), than I would prefer to not send any kernelupgrade to the current users cause support will be a nightmare (e.g. why does it not work anymore? it worked just before I did upgrade! I don't do any upgrades anymore cause you break things). If there aren't any noticeable side-effects than we can switch between both sorts of mali driver whenever RK messes one of them up... :P

igorpecovnik, 2018-05-31:
The problem is that "Linux mali" never worked properly while "Android mali" was broken recently. It was broken to the state that building fails. Experimenting caused switching to "Linux mali" ... but that one is not compatible with current mali libs ... bottom up. Goto my previous post :) 

We will not provide any kernel upgrade until this is solved.

Ntemis, 2018-05-31:
I would suggest one thing.
Follow&Use yocto rocko kernel update tag

igorpecovnik, 2018-05-31:
> I would suggest one thing.

We thought to implement this is as a source feature but so far things were eventually fixed upstream. I would not panic and wait 1-2 weeks and then proceed with alternative options.

Ntemis, 2018-05-31:
@chwe17 @Tonymac32 @igorpecovnik  basically is very easy to add support for mali hw acceleration i will explain the steps here and maybe someone with more knowledge of armbian building system surely can automate the steps.
1. Kernel headers -> linux-headers-rockchip
2. we need libmali-rk-dev_1.6-1_armhf.deb and libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb (r1p0 for Miqi) from https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf/libmali
3. Libdrm from https://github.com/rockchip-linux/libdrm-rockchip.git
- something like 
./autogen.sh --disable-intel --enable-rockchip-experimental-api
$ make && sudo make install 
will do the work :)
4.Libmali Library 
https://github.com/rockchip-linux/libmali
 Builds with:
$ cmake CMakeLists.txt
$ make 
5 https://github.com/rockchip-linux/mpp.git
 Builds with: 
$cmake -DRKPLATFORM=ON -DHAVE_DRM=ON
$make
6.And last wayland
from git://anongit.freedesktop.org/wayland/wayland
Build with:
./autogen.sh --disable-documentation
$ make && make install

teacupx, 2018-05-31:
Hi. JMCC here.

IMO, I think it is convenient to distinguish here three different aspects of the problem:

1. Mali DDK being updated to r18p0 for linux libmali:The commit referenced above was done six months ago, and they are still offering r14p0 [in their libmali repo](https://github.com/rockchip-linux/libmali/tree/29mirror/lib/arm-linux-gnueabihf). So, if the Android driver is now the only one compatible with r14p0, I understand they expect us to stay using it for linux too, until a r18p0 library is uploaded. 

This issue will come up with every update of the kernel driver or the userspace lib, and will necessarily cause GPU accel to be broken from time to time in the dev images. As a solution, it only requires that the kernel and lib maintainers at Armbian coordinate both ends before pushing a release.

2. The mali driver being broken now. I think the wisest is to freeze the kernel to a working commit, and unfreeze it when the Rockchip engineers fix it, if it is not too much trouble. Or, otherwise, as Igor says, do nothing and wait.

3. The issue with the game emulator: @Ntemis I don't know if you are aware that some developers and community members have been working in a full 3D & video implementation for the Armbian desktop in RK3288 boards , that is currently working quite well. I invite you to check it out [in this thread](https://forum.armbian.com/topic/7262-rk3288-media-testing-script/). All your contributions and suggestions are very welcome, so we can join efforts instead of spreading and duplicating our work.

teacupx, 2018-05-31:
@Ntemis : In a second read, it looks like you are developing an emulator-oriented distro based on Debian, correct? Well, in that case, it is very likely that the best solution for an X accelerated desktop will not be the same as for a game-only distro. For example, our solution differs substantially from the one LibreELEC currently uses.
One of the good things about Armbian is that it provides a very easy to use kernel config tool. I had no experience in kernel development, and I was able to work with it just in a couple hours. I suggest that you can use it to make a kernel that best tailors your needs, because probably some of the features that are included in the standard Armbian will be an unnecesary burden for your distro, and some features you need will be missing because they conflict with others that Armbian is using (that can be the case for the mali driver). Igor put a lot of work into that masterpiece that is the Armbian build script, let us take advantage of it ;)

Tonymac32, 2018-05-31:
@Miouyouyou does MiQi and Tinker require different libmali packages?  

As far as the commit reference, @teacupx, yes, that's why we aren't using it, there were some bugfixes that actually mattered that came in after that, I've been waiting on the next tagged release to freeze it.

teacupx, 2018-05-31:
> does MiQi and Tinker require different libmali packages?
According to [this](https://github.com/rockchip-linux/rk-rootfs-build/blob/933ab528b836b1e14d620f94d2bffa3055deb59b/overlay/etc/init.d/rockchip.sh#L19), the r1p0 lib is for RK3288w (a different revision of the SoC).
AFAIK, MiQi uses RK3288 as TinkerBoard. Several people reported in the Armbian thread that the same lib for the Tinker worked in the MiQi. Plus, according to [this](https://www.cnx-software.com/2017/12/07/rockchip-rk3288-to-be-phased-out-soon-gpd-xd-android-console-to-get-mediatek-mt8176-soc-4gb-ram-upgrade/#comment-550018), the two SoC's are not software compatible, and they would require changes in the kernel.

Miouyouyou, 2018-05-31:
I remember hearing something like this but I never knew the basis. With DRM/KMS/EGL, I never saw any difference, but I never tested with X11 back then (since you need the special snowflake Rockchip X11 server).

Miouyouyou, 2018-05-31:
Just tested by switching the libMali.so back and forth and, indeed, the r14p0-r1p0 is throwing a tantrum on my MiQi. The r0p0 works fine though. 

```
ERROR: The DDK is not compatible with any of the Mali GPUs on the system.
The DDK was built for 0x750 r1p0 status range [0..15], but none of the GPUs matched:
/dev/mali0 detected as 0x750 r0p0 status 1
Error: eglInitialize() failed with error: 0x3001
```

Ntemis, 2018-05-31:
@Miouyouyou then you might have a developers edition MiQi because mine is incompatible with r14p0-r0p0

Miouyouyou, 2018-05-31:
There's indeed multiple versions of MiQi devices, which don't make testing easier.
One with 2GB of RAM and one with 1GB of RAM.

I got the one with the 2GB and the green LED.

I also had one with a blue LED but it fried up :disappointed:

Ntemis, 2018-05-31:
Ah mine is the blue led now i see :)

Tonymac32, 2018-05-31:
hmmm, mine is blue, let me see.

Miouyouyou, 2018-05-31:
Well, I found that the Mali driver exports the GPU informations through sysfs :

cat /sys/devices/platform/ffa30000.gpu/misc/mali0/device/gpuinfo

```bash
root@miqi:/sys/devices/platform/ffa30000.gpu/misc/mali0# cat device/gpuinfo 
Mali-T76x 4 cores r0p0 0x0750
```

So there might a way to check this information easily :smiley_cat:

Tonymac32, 2018-05-31:
indeed, I just yanked it through the dmesg output, but yeah, r1p0.  That's it, EOL-ing the MiQi.   (J/K, but ffs)

Miouyouyou, 2018-05-31:
So Blue LED = RK3288W then ?

Ntemis, 2018-05-31:
@Tonymac32 why? all released miqis are blue led 2gb and r1p0

Tonymac32, 2018-05-31:
If so there's no mark on it.  The Tinker is an RK3288C, clearly marked.  

https://www.armbian.com/wp-content/uploads/2018/03/miqi-1.png  is mine, it's an angled shot, but no discerning marks.

@Ntemis it was a joke.  @teacupx can do a quick check in the script if he is so inclined to handle the difference.

Miouyouyou, 2018-05-31:
Since the mali driver depends on only one file : `libMali.so` . It might be possible to store in /usr/lib/... `libMali-r0p0.so` and `libMali-r1p0.so` then create a `libMali.so` symlink to the most appropriate one depending on the sysfs check.

Ntemis, 2018-05-31:
Yeap that would do just fine good call

Miouyouyou, 2018-05-31:
Or just trigger the right Rockchip libmali package installation on first boot. That way, it won't break the wonderful Rockchip libmali packages.

Ntemis, 2018-05-31:
what ever rocks your world, then ship it!

Miouyouyou, 2018-05-31:
:shipit:

Ntemis, 2018-05-31:
btw r18p0 has no KMS/DRM support only X11

Miouyouyou, 2018-05-31:
No gbm symbol at all ? That shucks. Does it have any vk symbol ?

`readelf -a libMali-of-the-r18p0.so | grep -i vk`

Also, where did you get a r18p0 user-space driver ? I don't see it on libmali Rockchip repo and don't see it ARM website either.

teacupx, 2018-05-31:
@Tonymac32 Check [here](https://www.cnx-software.com/2017/12/07/rockchip-rk3288-to-be-phased-out-soon-gpd-xd-android-console-to-get-mediatek-mt8176-soc-4gb-ram-upgrade/#comment-550018). It says that if the long alphanumeric row at the bottom left ends in "W", then it is the W version. That seems to be the case for the picture you sent.

Plus, in the same link, it states that they require different software, but the latest SDKs from Rockchip support both. However, we must take that into account just in case in the future some random MiQi's start to show problems and others don't: we'll have to add to the release notes "check the LED color" 🗡 

I can figure out some way of supporting both revisions in the scripts, but to be honest, I don't feel much like doing it if I don't have a MiQi myself. It is a pain to have others test before releasing. And I'm not going to buy one, so the only chance is that the board makers feel interested and send me a sample.

Ntemis, 2018-05-31:
ah sorry bud i did a mistake cause was some days ago and forgot the number and got curried away with all the r18 reference :)
Was talking about r22p0 actually not r18p0 and got them from here
https://chromium.googlesource.com/chromiumos/overlays/board-overlays.git/+/ef521b7f640511430e104ae2e46c148e34ebb868

Miouyouyou, 2018-05-31:
Well, the fbdev version of the Firefly drivers provided by ARM have Vulkan symbols.

So far I was able to list Vulkan "devices" and the queues properties, using these drivers. But I didn't try the display initialization yet. I'll have to focus on that part... when I got a free week-end or something.

The tutorial I'm following : https://software.intel.com/en-us/articles/api-without-secrets-introduction-to-vulkan-part-2
And some bits of code ... That might compile or not : https://github.com/Miouyouyou/Vulkan_Steps/tree/master
The driver  you want to test it with : https://developer.arm.com/products/software/mali-drivers/user-space -> Firefly -> r12p0 with fbdev support.

Tonymac32, 2018-05-31:
I see some familiar actors in that conversation.  Interesting, and seemingly completely unnecessary, probably added cost.

Miouyouyou, 2018-05-31:
Indeed ! Meanwhile, for the MiQi Red/Green issue, the fastest way to deal with this would be to check the rXpX version from sysfs and install the right package on first boot, IMHO.

The thing is, the rXpX issue might also be present with some Chromebooks too.

teacupx, 2018-05-31:
[People reported that the solution we are using in the Armbian script works in mainline MiQi](https://forum.armbian.com/topic/6506-tutorial-3d-video-acceleration-and-opencl-in-rk3288-boards-with-new-44-default-kernel/?do=findComment&comment=51024). I guess it was the plain RK3288 version, though.

Ntemis, 2018-05-31:
yes patched mainline kernel from rockchip has r14p0 support build-in

dmschwartzii, 2018-05-31:
@Ntemis yeah but is Bluetooth and sound stable?

Tonymac32, 2018-05-31:
My patches are @Miouyouyou 's patches for mainline.  
@dmschwartzii Bluetooth was recently resolved, Armbian has it rolled out, I've been playing music on a bluetooth speaker all day, actually.  HDMI sound quit on all three builds, that needs understood, if I roll back the scripts making the onboard USB codec work then there is no audio jack sound in desktop.  The DAI cells and codec include issues in 4.14 have been patched upstream.

dmschwartzii, 2018-05-31:
@Tonymac32 so Bluetooth works? HDMI sound does not?

Tonymac32, 2018-06-01:
Ja/Si/Oui/Da/Yes, I will be working on HDMI sound tonight, I've had enough Amlogic GXB bootloader witchcraft for a bit (how to magic the blobs into a working image with the mainline U-boot)

dmschwartzii, 2018-06-01:
@Tonymac32 keep me posted id like to test if you get it working

Tonymac32, 2018-06-01:
Another big round of commits hit 16 hours ago, let's take a look.  On the bright side, an improved Wifi driver is included.  I was waiting for it to hit Rockchip, it was an almost 20 MB patch

[5 more replies]