armbian/build #988: Failed to boot the system for orangepi lite

MS1987, 2018-05-31:
Hi. I just compile the armbian, for my orangepi Lite , using the methods as following:

> apt-get -y install git
git clone https://github.com/armbian/build
cd build
./compile.sh

At ui configure:
1. Choose "Full OS image for flashing"
2. Choose "Show a kernel configuration menu before compilation"
3. Choose "orangepilite"
4. Choose "default Vendor provided / legacy (3.4.x - 4.4.x)"
5. Choose "jessie Debian 8 Jessie"
6. Choose "Image with cosole interface (server)"

At linux menuconfig ui, no change, just exit.

Compile successfully. 
However, after I write the image to my sdcard and boot the orangepiLite. It failed to boot and always print the information as follows:

> Loading init Ramdisk from Legacy Image at 43300000
Image Name:   uInitrd
Image Type:   ARM Linux RAMDisk Image (gzip compressed)
Data Size:    3814305 Bytes = 3.6 MiB
Load Address: 00000000
Entry Point:  00000000
Verifying Checksum  OK
Starting kernel 

[sun8i_fixup]: From boot, get meminfo:
Start:  0x40000000
Size:   512MB
ion_carveout reserve: 160m@0 256m@0 130m@1 200m@1
ion_reserve_select: ion chipid  [0x2c00081!
ion_reserve_common: ion reserve: [0x56000000, 0x60000000]!
Loading, please wait...
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
done.
Begin: Will now check root file system ... fsck from util-linux 2.25.2
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p1] fsck.ext4 -a -C0 /dev/mmcblk0p1 
/dev/mmcblk0p1 has unsupported feature(s): metadata_csum
e2fsck: Get a newer version of e2fsck!
fsck exited with status code 8
done.
Warning: File system check failed but did not detect errors
mount: Invalid argument
done.
Target filesystem doesn't have requested /sbin/init.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... mount: No such file or directory
done.
Could not copy file: No such file or directory
No init found. Try passing init= bootarg.
Rebooting automatically due to panic= boot argument

I guess there is problem with rootfs, please help.

igorpecovnik, 2018-05-31:
Problems are only when booting into the old kernel. Strange.

MS1987, 2018-05-31:
@igorpecovnik , old kernel? You mean the kernel version I use is too old? Which version should I use?

Here is the same problem as mine:
https://forum.armbian.com/topic/4443-armbian-legacy-kernel-built-with-github-framework-is-not-booting/

igorpecovnik, 2018-05-31:
> You mean the kernel version I use is too old? Which version should I use?

It should be possible to build an image with old (3.4.y default) kernel while NEXT (4.14.y) works ...

zador-blood-stained, 2018-05-31:
These problems must be related to the Bionic build host, we need to adjust ext4 FS creation options.

igorpecovnik, 2018-05-31:
It works after this:
https://github.com/armbian/build/commit/7f07699c45a3ace165a238f37e7151dbea35d5d0
Boot log: http://ix.io/1bX5

Closing for now.

closed 2018-05-31 by igorpecovnik

MS1987, 2018-06-01:
@igorpecovnik , yes, it works with 4.14.y, but I want to confirm the different of this two version? Which one is more stable?

igorpecovnik, 2018-06-01:
It works with both kernels now. Well, it's hard to say. Some video related functions are missing in 4.14.y and it's not very well tested yet. People experienced years of uptime with the old kernel, while new is not that old yet.

vtrandal, 2018-11-25:
> It works after this:
> [7f07699](https://github.com/armbian/build/commit/7f07699c45a3ace165a238f37e7151dbea35d5d0)
> Boot log: http://ix.io/1bX5
> 
> Closing for now.
https://github.com/armbian/build/issues/988#issuecomment-393631435

> It works after this:
> [7f07699](https://github.com/armbian/build/commit/7f07699c45a3ace165a238f37e7151dbea35d5d0)
> Boot log: http://ix.io/1bX5
> 
> Closing for now.

Very nice work shown in the patch. I would like to apply that patch automatically but the line numbers don't match my version. Maybe I can make the edits manually or is there an easier way for more complicated patches in the future? I know my question is getting off topic and is more about patches (assuming this patch fixes my problem i.e. assuming it's same problem).