armbian/build #999: Can not modify the source code.

MS1987, 2018-06-04:
I want to modify something to the linux source code, if use the default config file, it always cover my code with the github one . So I add :

> IGNORE_UPDATES="yes"

 to the config-default.conf file, however, it always have the compile error:

> arch/arm/mach-sunxi/sun8i.c:201:41: error: redefinition of 'bmp085_i2c_bdi'

And I check the sun8i.c file, find that it's really redefine. So I remove the redefine part and compile again, however, it had the same error, and the sun8i.c recovered to the the version before I modified.
So it seemed that it is modified by the script, as I checked the log message, found :

> Cleaning [ linux-sun8i/sun8i 

So maybe here, it clean the sun8i source code ,and cover them with the default one, I want to know how to avoid it. Thank you.

igorpecovnik, 2018-06-04:
Use CREATE_PATCHES="yes" parameter.

closed 2018-07-02 by igorpecovnik

tuxsky, 2019-06-26:
I need to add extra souce code file c for a driver. In additoon to add CREATE_PATCHES="yes" flag, which is the steps to Made changes to Linux source code? Any tutorial por reference?

tuxsky, 2019-06-26:
I need to add extra souce code file c for a driver. In additoon to add CREATE_PATCHES="yes" flag, which is the steps to made changes to Linux source code? Any tutorial or reference?
As well I been following steps from https://docs.armbian.com/Developer-Guide_Build-Preparation/ and execute ./compile.sh BOARD=rk399 but script seems ignore my option, it build all targets but I want rk3399 only. Thank you

igorpecovnik, 2019-06-26:
RK3399 is not a board, but kernel family ... it can be "nanopim4" or "nanopct4" ... You can add CREATE_PATCHES="yes" to config-default and proceed compilation in interactive mode. In case you have more questions and you can't find answers in the manual, rather use forum: https://forum.armbian.com/forum/12-armbian-build-framework/