armbian/build #1007: Cubox u-boot dev branch won't boot from eMMC

pulento, 2018-06-07:
I have lots of HummingBoards 2 Edge with eMMC but now got a batch with the new SOM 1.5 and eMMC the original u-boot from next and stable branch (2013-10rc4) get stuck on SPL from time to time on the new SOMs so I'm trying the dev branch which has this new patch:

https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-cubox-next/U-Boot-mx6cuboxi-add-support-for-detecting-Revision-1.5-SoMs.patch

Copying SPL and u-boot.img to eMMC and booting produce the following:

```
U-Boot SPL 2018.03-armbian (Jun 07 2018 - 16:06:14 -0400)
Trying to boot from MMC1
Card did not respond to voltage select!
spl: mmc init failed with error: -95
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```
But if I have a SD card which has the old u-boot:

```
U-Boot SPL 2018.03-armbian (Jun 07 2018 - 16:06:14 -0400)
Trying to boot from MMC1

U-Boot 2013.10-rc4 (May 11 2018 - 13:06:31)

CPU:   Freescale i.MX6DL rev1.3 at 792 MHz
Reset cause: POR
Board: MX6-HummingBoard2
DRAM:  1 GiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   FEC [PRIME]
Hit any key to stop autoboot:  0
HummingBoard2 U-Boot >
```

And everything goes fine. So it seems that the SPL tries MMC1 which I suppose is SD card so when no card is inserted it get stuck. 

How can I force MMC0 inside the code or see if the new SPL honours the HummingBoards jumpers.

TIA

igorpecovnik, 2018-06-07:
I am sorry but: 

- none of the developers have a board with eMMC so this function was never observed nor tested. 
- our general policy says that we don't provide technical support for development versions. Too expensive.
- this is build script issue tracker. Please start a discussion in the [forum.armbian.com](https://forum.armbian.com) if necessary.

closed 2018-06-07 by igorpecovnik

pulento, 2018-06-07:
I understand. The stable and next branches won't pass SPL on SOM 1.5 with eMMC sometimes (about 1/3).

I wan't to help on this issue I'm compiling both branches just need to test if I force MMC2_2 on SPL it works form eMMC. I'll continue on forums.

Thanks