armbian/build #1022: Enable more LED triggers on RK3328 (ROCK64)

markh-de, 2018-06-25:
Can we please have more LED triggers enabled for RK3328 (ROCK64)?

For example, I find it very useful to have a timer trigger for error condition reporting, or a heartbeat, on the ROCK64's rather dim red LED for headless systems (the white LED is used with trigger 'mmc1').

I'd suggest to enable the same LED triggers that we usually have on e.g. sunxi, i.e. all of the available LED triggers set to either "y" or "m".

Thanks for your consideration. Should I create a PR, if granted?

igorpecovnik, 2018-06-25:
> Thanks for your consideration. Should I create a PR, if granted?

I don't mind if it works as expected.

markh-de, 2018-06-26:
Fixed with PR #1023.

closed 2018-06-26 by markh-de