armbian/build #1033: Improper replacement of log.hdd causing failure with logrotate

rabits, 2018-07-05:
In the **Armbian_5.50.180630_Tinkerboard_Ubuntu_bionic_next_4.14.52.img** I found a bug with zram logrotating:

Seems that cause file is [packages/bsp/common/usr/lib/armbian/armbian-hardware-optimization#L47](/armbian/build/commit/75e17b0e21fac1b05bd9ab84c4e479316e38c957#diff-9783ace04ab7bab2c8b6a00488da9146R47) contains `sed` that replaces `log` to `log.hdd` in the logrotate configuration, that also affects target file `/etc/logrotate.d/rsyslog` and replaces valid `postrotate` action `/usr/lib/rsyslog/rsyslog-rotate` to invalid `/usr/lib/rsyslog.hdd/rsyslog-rotate` that doesn't exist.

Error:
```
$ sudo /usr/lib/armbian/armbian-truncate-logs
logrotate_script: 2: logrotate_script: /usr/lib/rsyslog.hdd/rsyslog-rotate: not found
...
```

BTW seems that logratate can't affect journald and it will consume all the available disk space on the zram disk.

igorpecovnik, 2018-07-06:
Thanks. This should fix the issue while for journald we have to figure out something.

rabits, 2018-07-06:
Sure, I will test it

tosiara, 2018-09-19:
FWIW, I have just got this email from an Orange Pi Lite running Armbian_5.59_Orangepilite_Ubuntu_bionic_next_4.14.65:
```
/etc/cron.daily/logrotate:
logrotate_script: 2: logrotate_script: /usr/lib/rsyslog.hdd/rsyslog-rotate: not found
error: error running non-shared postrotate script for /var/log.hdd/syslog of '/var/log.hdd/syslog
'
run-parts: /etc/cron.daily/logrotate exited with return code 1
```
I flashed the board on Monday and was playing with settings which resulted in a situation when I had to reinstall rsyslog package.

If sed has been already fixed, do you know what else could trigger sed command to replace "log" with "log.hdd"?

tosiara, 2019-04-18:
Armbian_5.75_Orangepilite_Ubuntu_bionic_next_4.19.20
Still getting those emails:
```
/etc/cron.daily/logrotate:
logrotate_script: 2: logrotate_script: /usr/lib/rsyslog.hdd/rsyslog-rotate: not found
error: error running non-shared postrotate script for /var/log.hdd/syslog of '/var/log.hdd/syslog
'
run-parts: /etc/cron.daily/logrotate exited with return code 1
```

Stefan300381, 2019-05-02:
same here: 

```
/etc/cron.daily/logrotate:
logrotate_script: 2: logrotate_script: /usr/lib/rsyslog.hdd/rsyslog-rotate: not found
error: error running non-shared postrotate script for /var/log.hdd/syslog of '/var/log.hdd/syslog
'
run-parts: /etc/cron.daily/logrotate exited with return code 1
```

**/etc/armbian-release**
```
BOARD=orangepiplus2e
BOARD_NAME="Orange Pi+ 2E"
BOARDFAMILY=sun8i
VERSION=5.75
LINUXFAMILY=sunxi
BRANCH=next
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage
```

**uname -a**
`Linux orangepiplus2e 4.19.25-sunxi #5.78 SMP Mon Apr 8 07:46:22 CEST 2019 armv7l armv7l armv7l GNU/Linux
`

danger89, 2019-05-06:
Yep. Please reopen:

```
/etc/cron.daily/logrotate: logrotate_script: 2: logrotate_script: /usr/lib/rsyslog.hdd/rsyslog-rotate: not found error: error running non-shared postrotate script for /var/log.hdd/syslog of '/var/log.hdd/syslog ' run-parts: /etc/cron.daily/logrotate exited with return code 1
```

`Linux bananapi 4.19.25-sunxi #5.78 SMP Mon Apr 8 07:46:22 CEST 2019 armv7l armv7l armv7l GNU/Linux`

danger89, 2019-05-06:
Open the following config file (eg. via nano):
`nano /etc/logrotate.d/rsyslog`

Changing all the lines with:
`/usr/lib/rsyslog.hdd/rsyslog-rotate`
To:
` /usr/lib/rsyslog/rsyslog-rotate`

Hopefully solved the problem..!?

danger89, 2019-05-10:
Actually that solved the problem, thanks past me!

However, now the following issue pop-ups:
`error: error running shared postrotate script for '/var/log.hdd/nginx/*.log '`

Stefan300381, 2019-05-20:
it solved the problem for me as well, but it came back (probably after a reboot of the machine)

Stefan300381, 2019-08-19:
problem still exists in:

Linux orangepiplus2e 4.19.62-sunxi #5.92 SMP Wed Jul 31 22:07:23 CEST 2019 armv7l armv7l armv7l GNU/Linux

can we reopen this ticket?

igorpecovnik, 2019-10-09:
@Stefan300381 Perhaps this https://github.com/armbian/build/pull/1588 solved it?

Stefan300381, 2019-10-10:
@igorpecovnik 
tbh I'm not a github expert and thus I don't know what has been changed where. Anyway, I ran a dist-upgrade on my OrangePi Plus 2E and I'm still on 5.90

Let's see tomorrow morning

igorpecovnik, 2019-10-10:
This code change is for now only on Github, You need to build an image from sources or implement manually to see if this problem was fixed -> https://docs.armbian.com/Developer-Guide_Build-Preparation/

Stefan300381, 2019-10-10:
alright, thanks. I'll wait then until apt upgrade solves the issue :-)

igorpecovnik, 2020-01-07:
Unable to reproduce, closing.

closed 2020-01-07 by igorpecovnik