armbian/build #1034: Sid and Testing debian repos [proposed enhacement]

78bash, 2018-07-06:
Switching to sid or testing in the sources.list doesn't currently work.

igorpecovnik, 2018-09-26:
We have no resources to do this, but if you like, here is where you need to look:

https://github.com/armbian/build/blob/master/lib/build-all.sh#L18
https://github.com/armbian/build/blob/master/lib/desktop.sh#L89
https://github.com/armbian/build/blob/master/lib/distributions.sh#L265-L348
https://github.com/armbian/build/blob/master/lib/distributions.sh#L265-L348
https://github.com/armbian/build/blob/master/lib/general.sh#L345-L352
https://github.com/armbian/build/blob/master/lib/main.sh#L198-L201
https://github.com/armbian/build/blob/master/lib/makeboarddeb.sh#L244-L289
https://github.com/armbian/build/blob/master/lib/repository.sh#L26
And entire section:
https://github.com/armbian/build/tree/master/packages/extras-buildpkgs

lanefu, 2019-05-30:
closing stale issue

closed 2019-05-30 by lanefu

EvilOlaf, 2019-05-30:
Buster has been added while in testing stage.

igorpecovnik, 2019-05-30:
Indeed. Buster boots and works with minor issues, but a bulk of problems are expected to be found here: https://github.com/armbian/config