armbian/build #1035: nand-sata-install doesn't seem to work for nanopc-T3

daiaji, 2018-07-07:
I used the image of [nanopi M3](https://www.armbian.com/nanopi-m3/)
After using nand-sata-install, my nanopc-T3 could not boot.

igorpecovnik, 2018-07-07:
It's a known problem of T3 eMMc support within u-boot. The script works properly, but until someone with a better understanding of u-boot solve this, it will remain broken.

daiaji, 2018-07-07:
@igorpecovnik Does this mean that nanopc-T3+ will also have such a problem?
Friendlyarm seems to have given up technical support for Nanopc-T3, which is too bad.😭

igorpecovnik, 2018-07-07:
> Does this mean that nanopc-T3+ will also have such a problem?
Yes.

>Friendlyarm seems to have given up technical support for Nanopc-T3, which is too bad.

Board from FA perspective works well enough. If you want more and better support, you need to throw money into the R&D. We already spent hundreds of engineering hours solo on this board and all we got back are complains and user demands. There is virtually no financial help except our own pocket which is closed for further investment. If you want some feature fixed, covering development time might be a possible way to go.

daiaji, 2018-09-04:
@igorpecovnik Some reports on these days.
I've been trying to compile Linux 4.14.39.
But this version of the kernel is not working on Nanopc-T3.
Until I found that rafaello7 provided bl1-nanopi-m3 1.2.
The update log reads "Added correction for kernel 4.14.39+".
That solved my problem.
At least I can make kernel 4.14.39 work on EMMC.😀

igorpecovnik, 2018-09-04:
Yes, I know that. Armbian is aligned with this + a few patches on top, [4.14.65](http://ix.io/1kMg) ATM but have you succeeded in booting from eMMC? That is still unresolved.

daiaji, 2018-09-04:
@igorpecovnik I didn't have Linux 4.14.65, so I didn't test.
This may require more patches to u-boot.
But at least Linux 4.14.39 can now work in EMMC. 😁

igorpecovnik, 2018-09-04:
The problem is u-boot. I am aware of that. I already tried to merge things in but failed. Which u-boot do you use?

daiaji, 2018-09-04:
https://github.com/rafaello7/u-boot-nanopi-m3/ and https://github.com/rafaello7/bl1-nanopi-m3
Use https://github.com/rafaello7/nanopi-boot-tools brush to write u-boot
https://github.com/rafaello7/debian-installer-nanopi-m3
😀

igorpecovnik, 2018-09-04:
Thank you. Then I have to check what we did wrong.

daiaji, 2018-09-05:
@igorpecovnik I tried to upgrade the kernel to 4.14.65 version and compile it through patches.
He can work on the u-boot provided by rafaello7.
So why not be a bit more crazy?
So I tried to update the kernel to version 4.18.
It seems that https://github.com/rafaello7/linux-nanopi-m3-v4.14 lacks some ACPI source code in the process, but I don't think it's fatal
Until updated to version 4.18, it seems that patch is in conflict with some of the source code provided by Samsung.
It seems that I should try to solve this problem tomorrow.
Maybe it's not difficult.

igorpecovnik, 2018-09-05:
I already tried to go up to 4.16.y a while ago but noticed that a lot of things has been changed regarding handling CPU. It's above my understanding. I have Fire3 and T3+ now on my table and I already fix few minor things, have found a fix for eMMC booting https://github.com/rafaello7/debian-installer-nanopi-m3/blob/master/nanopi-bootable/debian/postinst#L80-L84 and I hope it will push commits tomorrow.

igorpecovnik, 2018-09-06:
https://github.com/armbian/build/commit/86a00ada34324d902dca380ccdf06d935bf095f9

closed 2018-09-06 by igorpecovnik