armbian/build #1042: armbian-add-overlay error

roderick1980, 2018-07-13:
Hello,
according to "uname -a", I've got a Linux olimex 4.15.0-sunxi system.
I've successfully installed the following kernel headers: linux-headers-dev-sunxi and also linux-headers-next-sunxi. 

When I execute the "armbian-add-overlay" command from /boot/overlay-users, I've got the next error message:
"Kernel headers are not installed properly. Please install the kernel headers package"

Can anybody help me with this?

Thanks in advance

igorpecovnik, 2018-07-13:
Use NEXT branch, kernel 4.14.y ... and you will have overlays. You are using some DEVelopment version, where patches for overlay support might be missing. 

You can try 4.17.y, where this should work, but ... it's also in a testing phase: https://forum.armbian.com/topic/7503-we-need-your-help/

kitasaran, 2018-07-15:
[logs-14_07_2018-18_28_59.zip](https://github.com/armbian/build/files/2195436/logs-14_07_2018-18_28_59.zip)
It is my log

roderick1980, 2018-07-16:
Hi again,
I've tried with sunxi-4.18, but unfortunately I get the following command line output when I execute "compile.sh":

_...
 CC      lib/strto.o
  CC      lib/strmhz.o
  LD      lib/built-in.o
  CC      examples/standalone/stubs.o
  CC      examples/standalone/hello_world.o
  LD      examples/standalone/libstubs.o
  LD      examples/standalone/hello_world
  OBJCOPY examples/standalone/hello_world.srec
  OBJCOPY examples/standalone/hello_world.bin
  LD      u-boot
board/sunxi/built-in.o: In function `scsi_init':
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/board/sunxi/ahci.c:79: undefined reference to `ahci_init'
cmd/built-in.o: In function `do_scsi':
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/cmd/scsi.c:35: undefined reference to `scsi_bus_reset'
drivers/built-in.o: In function `scsi_write':
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/drivers/scsi/scsi.c:265: undefined reference to `scsi_exec'
drivers/built-in.o: In function `scsi_read':
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/drivers/scsi/scsi.c:201: undefined reference to `scsi_exec'
drivers/built-in.o: In function `scsi_detect_dev':
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/drivers/scsi/scsi.c:503: undefined reference to `scsi_exec'
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/drivers/scsi/scsi.c:534: undefined reference to `scsi_exec'
drivers/built-in.o: In function `scsi_read_capacity':
/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/drivers/scsi/scsi.c:372: undefined reference to `scsi_exec'
drivers/built-in.o:/home/rlopez/Work/som204/build/cache/sources/u-boot/v2018.05/drivers/scsi/scsi.c:397: more undefined references to `scsi_exec' follow
/home/rlopez/Work/som204/build/cache/toolchains/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ld.bfd: BFD (Linaro_Binutils-2017.11) 2.28.2.20170706 assertion fail /home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/arm-linux-gnueabihf/snapshots/binutils-gdb.git~users~linaro~binutils-2_28-branch/bfd/elf32-arm.c:9514
/home/rlopez/Work/som204/build/cache/toolchains/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ld.bfd: BFD (Linaro_Binutils-2017.11) 2.28.2.20170706 assertion fail /home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/arm-linux-gnueabihf/snapshots/binutils-gdb.git~users~linaro~binutils-2_28-branch/bfd/elf32-arm.c:9514
/home/rlopez/Work/som204/build/cache/toolchains/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ld.bfd: BFD (Linaro_Binutils-2017.11) 2.28.2.20170706 assertion fail /home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/arm-linux-gnueabihf/snapshots/binutils-gdb.git~users~linaro~binutils-2_28-branch/bfd/elf32-arm.c:9514
Makefile:1336: recipe for target 'u-boot' failed
make: *** [u-boot] Error 1
[ error ] ERROR in function compile_uboot [ compilation.sh:173 ]
[ error ] U-boot compilation failed 
[ o.k. ] Process terminated_ 

Any clue about this error?

Thanks in advance

Rodrigo

igorpecovnik, 2018-07-16:
What are you compiling, build logs?

roderick1980, 2018-07-16:
Hello Igor,

thanks for your interest.

You will find the URL of the logs below:

[output.log](https://github.com/armbian/build/files/2198105/output.log)
[patching.log](https://github.com/armbian/build/files/2198106/patching.log)

By the way, I've found that compiling for A20 instead of A20 som204 gives no error.

Kind regards

igorpecovnik, 2018-07-17:
@StefanMavrodiev Can you check up on this? I just switched master branch -> NEXT u-boot and kernel to 2018.05 / 4.17.y and som204 patch broke. I guess it was broken before on DEV as well.

StefanMavrodiev, 2018-07-18:
@igorpecovnik A20-SOM204 is supported in 2018.05, so you need to remove **add-A20-Olimex-SOM204-EVB-board.patch** and **board_olimex-som204-a20/Change-GMAC-phy-address.patch**.

roderick1980, 2018-07-18:
Hello,

Yesterday I tried to compile against a new Armbian branch called sunxi-4.18-tmp and I found an error while compiling u-boot. By comparing som204 config against a20-som config and looking at the compile errors, I assume that problem was in the lack of a CONFIG_SCSI_AHCI variable declaration. 

After inserting this config variable, I could succesfully build a linux 4.17 kernel based system from this branch and HDMI DDC functionality is now available as a /dev/i2c-X device.

I attach the patch that worked for me in case you find it of any use.

Thank you very much for your help.

[u-boot-sunxi-next.patch.zip](https://github.com/armbian/build/files/2204852/u-boot-sunxi-next.patch.zip)

igorpecovnik, 2018-07-18:
Try master now. It was fixed a few hours ago and it probably works.

closed 2021-06-14 by lanefu