armbian/build #1044: Does not work after assembling on the Asus Tinker Board (latest update from GitHub)

kitasaran, 2018-07-14:
Using Ubuntu 16.04 x64 host, build builds without errors

igorpecovnik, 2018-07-14:
Tinkerboard default kernel building might be broken ... but with so little data it's impossible to provide any assistance. We provide three different kernels, four different userspace ... 

Where are build logs?
Where are boot logs? (necessery for diagnose "doesn't work")

Tonymac32, 2018-07-14:
The "does not work" can be at least better defined from my side as Uboot is fine, loading kernel image goes without errors, heartbeat LED is active, but nothing on HDMI or serial from the kernel.  (tried UART2 and 3 in case of "losing" the uart3 patch).  I'm doing home repairs, so I've been slow to react to this.

Activity LED quits after about 3 minutes I see.  

 Unfortunately for everyone wanting fixes Summer means more sunshine and time to do outside work and enjoy nature...  So things move more slowly

igorpecovnik, 2018-07-14:
All day out here as well :) But NEXT 4.14.y kernel works, right?

Tonymac32, 2018-07-14:
yes, the current download does, I'll check building one later.

kitasaran, 2018-07-15:
[logs-14_07_2018-18_28_59.zip](https://github.com/armbian/build/files/2195438/logs-14_07_2018-18_28_59.zip)

kitasaran, 2018-07-15:
Sorry