armbian/build #1055: Tinkerboard basic overlay support

oliskoli, 2018-07-19:
Hi together and many thanks for your work!

I've made some patches to get basic overlay support into the next (!) branch. I have successfully tested common sensors like dht11 or onewire w1, but got also other stuff working as pps or a fbtft display (waveshare). These patches would be enough for me, but would need some work to get a full support (i.e. fixup script).

Are there interests on that ? And, if yes, do I have to go over git/fork/... or can I provide my "userpatch" directory ?

Kind regards

igorpecovnik, 2018-07-20:
Hi and thank you for your interest. Well, there is certainly general interest in having overlay support for Tinkerboard. When you have things working to some degree, that testing is possible and patches doesn't break anything, put them into patch/kernel/rockchip-next subdirectory and submit as PR to the master branch.

oliskoli, 2018-07-20:
Okay. Nice, i'll do it. Give me five minutes :-)

Tonymac32, 2019-02-06:
@oliskoli check the Next recipe, it includes overlays on kernel 4.19

closed 2019-02-06 by igorpecovnik