armbian/build #1067: sunxi-cir module does not work in kernels > 3.4

motaz9k, 2018-07-30:
With the kernel  3.4 lirc working great, but with each newer version it does not work at all

patient : orange pi zero ( [i have this set](https://www.banggood.com/3-in-1-Orange-Pi-Zero-512MB-Development-Board-Expansion-Board-Black-Case-Kit-p-1135881.html) )

 **uname -a**
`Linux orangepizero 4.18.0-rc6-sunxi #264 SMP Sat Jul 28 15:39:33 UTC 2018 armv7l armv7l armv7l GNU/Linux`

**sudo modprobe sunxi_cir**
**lsmod**
```
Module                  Size  Used by
sunxi_cir              16384  0
zram                   20480  4
sun8i_codec_analog     24576  0
pwrseq_simple          16384  1
snd_soc_core          106496  1 sun8i_codec_analog
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm                65536  2 snd_pcm_dmaengine,snd_soc_core
snd_timer              24576  1 snd_pcm
snd                    45056  3 snd_timer,snd_soc_core,snd_pcm
soundcore              16384  1 snd
cdc_acm                20480  2
sun4i_tcon             24576  0
sun8i_mixer            32768  0
uio_pdrv_genirq        16384  0
uio                    16384  1 uio_pdrv_genirq
cpufreq_dt             16384  0
thermal_sys            49152  1 cpufreq_dt

```

**sudo systemctl status lirc**
```
● lirc.service - Linux Infrared Remote Control
   Loaded: loaded (/usr/lib/systemd/system/lirc.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-07-30 15:19:55 CEST; 10min ago
 Main PID: 1020 (lircd)
   CGroup: /system.slice/lirc.service
           └─1020 /usr/sbin/lircd --nodaemon --driver=default --device=/dev/lirc0 --listen

Jul 30 15:19:56 orangepizero lircd-0.9.0[1020]: lircd(default) ready, using /var/run/lirc/lircd
Jul 30 15:22:53 orangepizero lircd[1020]: lircd-0.9.0[1020]: accepted new client from 192.168.1.100
Jul 30 15:22:53 orangepizero lircd[1020]: lircd-0.9.0[1020]: could not get file information for /dev/lirc0
Jul 30 15:22:53 orangepizero lircd[1020]: lircd-0.9.0[1020]: default_init(): No such file or directory
Jul 30 15:22:53 orangepizero lircd[1020]: lircd-0.9.0[1020]: Failed to initialize hardware
Jul 30 15:22:53 orangepizero lircd-0.9.0[1020]: accepted new client from 192.168.1.100
Jul 30 15:22:53 orangepizero lircd-0.9.0[1020]: could not get file information for /dev/lirc0
Jul 30 15:22:53 orangepizero lircd-0.9.0[1020]: default_init(): No such file or directory
Jul 30 15:22:53 orangepizero lircd-0.9.0[1020]: Failed to initialize hardware

```

**sudo systemctl restart lirc**
**sudo systemctl status lirc**
```
● lirc.service - Linux Infrared Remote Control
   Loaded: loaded (/usr/lib/systemd/system/lirc.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-07-30 15:32:22 CEST; 13s ago
 Main PID: 3187 (lircd)
   CGroup: /system.slice/lirc.service
           └─3187 /usr/sbin/lircd --nodaemon --driver=default --device=/dev/lirc0 --listen

Jul 30 15:32:22 orangepizero systemd[1]: Started Linux Infrared Remote Control.
Jul 30 15:32:22 orangepizero lircd[3187]: lircd-0.9.0[3187]: lircd(default) ready, using /var/run/lirc/lircd
Jul 30 15:32:22 orangepizero lircd-0.9.0[3187]: lircd(default) ready, using /var/run/lirc/lircd
```
 **sudo ls /dev/lirc***
`ls: cannot access '/dev/lirc*': No such file or directory`

**sudo cat /proc/bus/input/devices** is empty

igorpecovnik, 2018-07-30:
This is build script issue tracker. Report and discuss here:
https://groups.google.com/forum/#!forum/linux-sunxi
and/or
https://forum.armbian.com/forum/4-development/
Not everything works yet in a modern kernel:
https://linux-sunxi.org/Linux_mainlining_effort#Status_Matrix

closed 2018-07-30 by igorpecovnik

motaz9k, 2018-07-30:
I understand, it's a shame because I have to choose between ble support* and a working lirc module only in kernel 3.4 

_*bluez requires at least a kernel 3.14 to support bluetooth low energy_

motaz9k, 2018-07-30:
excuse me, I did not say at the beginning that I've tested all the builds

igorpecovnik, 2018-07-30:
Please use forum(s) and provide as much useful data as possible. Than things might get resolved.

Here is just not right place.