armbian/build #1072: nand-sata-install formating errors

spot62, 2018-08-07:
After  running nand-sata-install at the step
 "eMMC install: Formating /dev/mmcblk1 to ext4 ... please wait:"

```
mount: wrong fs type, bad option, bad superblock on /dev/mmcblk1p1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
```
dmesg:
```
[   72.516530] EXT4-fs (mmcblk1p1): Unrecognized mount option "flush" or missing value
[   73.928286] EXT4-fs (mmcblk1p1): Unrecognized mount option "flush" or missing value
[   74.238966] EXT4-fs (mmcblk1p1): Unrecognized mount option "flush" or missing value
[   74.822857] +mmc_blk_issue_discard_rq,1809
[   74.822921] mmc2:mmc_do_erase: erase from 8192 to 40959 arg 0x00000003
[   74.839657] -mmc_blk_issue_discard_rq,1845
[   74.839923] +mmc_blk_issue_discard_rq,1809
[   74.839975] mmc2:mmc_do_erase: erase from 40960 to 4235263 arg 0x00000003
[   75.908687] -mmc_blk_issue_discard_rq,1845
[   75.908733] +mmc_blk_issue_discard_rq,1809
[   75.908749] mmc2:mmc_do_erase: erase from 4235264 to 8429567 arg 0x00000003
[   77.046410] -mmc_blk_issue_discard_rq,1845
[   77.046454] +mmc_blk_issue_discard_rq,1809
[   77.046472] mmc2:mmc_do_erase: erase from 8429568 to 12623871 arg 0x00000003
[   78.106266] -mmc_blk_issue_discard_rq,1845
[   78.106311] +mmc_blk_issue_discard_rq,1809
[   78.106329] mmc2:mmc_do_erase: erase from 12623872 to 15117183 arg 0x00000003
[   78.738142] -mmc_blk_issue_discard_rq,1845
[   82.851026] EXT4-fs (mmcblk1p1): Unrecognized mount option "flush" or missing value
[   82.957791] EXT4-fs (mmcblk1p1): Unrecognized mount option "compress-force=zlib" or missing value
```
and my NanoPi Neo Air could not boot from eMMC

igorpecovnik, 2018-08-07:
This is build script issue tracker. 

Please use https://forum.armbian.com and provide data with our diagnostic tools (armbianmonitor -u) to narrow problems down. It looks like its a hardware failure, which you need to solve with board maker if you still have wareanty.

spot62, 2018-08-07:
Ок. Thank you.

closed 2018-08-07 by spot62