armbian/build #1080: Possible issue. Cusotm kernel and DTB files not installed?

g40, 2018-08-11:
Hello,

Using master from this repo and have a custom kernel repo which builds correctly for the NanoPi K1+ (H5), I find that the kernel .deb is missing the `Image` and all `dtb*` files. In which script are these files copied and how best to debug/warn if the copy fails?

The image boots correctly if I manually install the kernel and DTB(O) files onto the target SD card.

TAIA

igorpecovnik, 2018-08-11:
> and have a custom kernel repo

What custom kernel? You don't use the default kernel source and you don't use patches? It is normal that it doesn't pack properly.

g40, 2018-08-15:
Thanks. Hmm, O.K., Did not realise the install was done via patch.  Closing.

closed 2018-08-15 by g40