armbian/build #1081: Monitoring issues

ThomasKaiser, 2018-08-12:
Since refactoring the services the `stop` case in [armbian-hardware-monitor](https://github.com/armbian/build/blob/master/packages/bsp/common/usr/lib/armbian/armbian-hardware-monitor#L163) has become useless since never executed.

I tried to nail this down some time ago but to no avail. `systemd` NOOB still.

Related: To get nicer log output it would be great if `armbian-hardware-monitor` could wait for `firstrun` and `resize2fs` services to finish before starting to log. Also no idea how to achieve this.

lanefu, 2018-09-28:
@ThomasKaiser 

have any more details about what you mean by "since never executed" ?    

I've just starting digging through the scripts and the forum to understand their role and make sense of it all.

ThomasKaiser, 2018-09-28:
The `armbian-hardware-monitor` script has a start and a stop case. But the script will never be called on shutdown. I don't think it's needed to understand anything inside the script(s) since it boils down to how `systemd` handles such stuff (keeping the script active or not, executing it on shutdown/reboot).

sfx2000, 2018-10-02:
with systemd - calling init is calling systemd already - so the start/stop events in the script aren't particularly needed

lanefu, 2018-12-19:
Okay so according to the systemd.service docs, the unit file for the armbian-hardware-monitor service does the right thing and the shutdown cleanup steps should be happening.    

Since the job is a oneshot, and has the `RemainAfterExit` flag is true, the `ExecStop` command is supposed to be executed.

see example 3 https://www.freedesktop.org/software/systemd/man/systemd.service.html#Examples

Anyway that's the _should_.   

The reality is that I'm not seeing `###shutdown` logged anywhere in `/var/log/armbian-hardware-monitor.log` to indicate the stop command is actually happening...

I'm going to rename issue for clarity now that i understand it.