armbian/build #1093: Failed to build OrangePI PC+ image (uboot build issues)

ianzag, 2018-08-31:
Armbian branch is master
Build configuration options:

```
KERNEL_ONLY="yes"
KERNEL_CONFIGURE="no"
BUILD_DESKTOP="no"
BOARD="orangepipcplus"
BRANCH="next"
RELEASE="stretch"
```

Build fails with the following error:

```
  FDTGREP dts/dt-spl.dtb
  SHIPPED dts/dt.dtb
  CAT     u-boot-dtb.bin
  COPY    u-boot.dtb
  COPY    u-boot.bin
  MKIMAGE u-boot.img
./tools/mkimage: Can't map u-boot.img: Invalid argument
Makefile:1090: recipe for target 'u-boot.img' failed
make: *** [u-boot.img] Error 1
make: *** Waiting for unfinished jobs....
............
  OBJCOPY spl/u-boot-spl-nodtb.bin
  COPY    spl/u-boot-spl.bin
  MKSUNXI spl/sunxi-spl.bin
make: *** wait: No child processes.  Stop.
[error] ERROR in function compile_uboot [.[0;33m compilation.sh:173 .[0m]
[error] U-boot compilation failed
```
Looks like some u-boot issues. Can you please help us to build an image?
Thanks!

[orangepipcplus-build-logs.tar.gz](https://github.com/armbian/build/files/2339939/orangepipcplus-build-logs.tar.gz)

igorpecovnik, 2018-08-31:
Nothing wrong here. Try: 

apt purge u-boot-tools

apt install u-boot-tools

ianzag, 2018-08-31:
Igor thank you for reply!

Unfortunately, this does not help. I re-installed system wide u-boot-tools and removed all sources from cache. Build  still fails in the same place in u-boot. I guess Makefile uses its own built version of mkimage from ./tools/mkimage and does not use one installed in system. 

New build logs are attached. Thanks!

[orangepipcplus-build-logs.tar.gz](https://github.com/armbian/build/files/2340095/orangepipcplus-build-logs.tar.gz)

igorpecovnik, 2018-08-31:
Then it must be something related to Vagrant/Virtualbox. Strange. I don't use such setup so I can't do quick checking.

martinayotte, 2018-08-31:
I'm using VirtualBox to do some of my builds and I don't have such issue ...

zador-blood-stained, 2018-09-01:
```
## Build script directories
Build directory is located on:
TARGET   SOURCE  FSTYPE  AVAIL
/vagrant vagrant vboxsf 181.8G
```

Most likely this: https://www.virtualbox.org/ticket/819#comment:61

And [our Vagrant configuration](https://github.com/armbian/build/blob/master/Vagrantfile) specifically avoids running the build script from the `/vagrant` mount point, even though this makes it much harder to run the build script with local changes.

ianzag, 2018-09-03:
Hi all,

Yes, I was building from inside /vagrant mount point => looks like there's some problems with mmap/exec. When I switched build into let's say /home/root/armbian problem has gone (I can build u-boot with no problems). Though this behavior wasn't obvious for beginner in vagrant/orpi build environment.

Thanks! I think issue can be closed.

closed 2018-09-03 by igorpecovnik