armbian/build #1098: 'Wrong' permissions of /

ThomasKaiser, 2018-09-05:
(Recent?) Armbian images set permissions of the root directory to `777` while they should be `755`. As a result some software checking recursively for permissions like `sshd` complain and [refuse to work in specific modes](https://forum.openmediavault.org/index.php/Thread/23234-ssh-with-public-key-bad-ownership-or-modes-for-directory/?postID=179430#post179430).

Is this by intention?

igorpecovnik, 2018-09-05:
No, haven notice. Where is that set?

ThomasKaiser, 2018-09-05:
No idea where it originates from. But at least with latest images it's set that way. I realized it with Stretch and now double checked with Bionic:

    root@nanopik1plus:~# ls -lad /
    drwxrwxrwt 22 root root 4096 Aug 20 18:43 /
    root@nanopik1plus:~# lsb_release -c
    Codename:	bionic

Anyway, doing an `chmod 755 $rootfs-root` prior to closing the image should fix the issue.

closed 2018-09-05 by ThomasKaiser

adatum, 2019-08-24:
I ran into this issue on `OMV_4_Odroid_XU4_HC1_HC2.img.xz` release dated 2018-06-03 and downloaded on 2019-08-08 at https://sourceforge.net/projects/openmediavault/files/OMV%204.x%20for%20Single%20Board%20Computers/

ssh authentication failed (public key denied) when password authentication is disabled. It worked fine with password authentication enabled.

The Authentication System Log in openmediavault showed 
> sshd: Authentication refused: bad ownership or modes for directory /

`chmod 755 /` fixes the problem.

sw25481, 2020-09-21:
I ran into the same problem with openmediavault_5.3.9-amd64.iso downloaded 20 Sep 2020 installed into a VMware virtual machine.  chmod 755 / fixed if for me as well

Herjar, 2021-04-15:
I got the same issue with `openmediavault_5.5.11-amd64.iso` installed as vm in proxmox. `sudo chmod 755 /` fixed it here too.

lededev, 2021-04-21:
armbian have make /root directory dr-xr-xr-x, and with wrong user id.
That cause ssh pub key login failed.
fix: 
chmod 700 /root
chown root:root /root

igorpecovnik, 2021-04-21:
Stock Armbian doesn't seem to have this problem:
``
drwx------   7 root root  4096 Apr 21 00:35 root
``

lededev, 2021-04-21:
Well, if not stock, that may caused by some programs.

commit 46340a24 (2018-09-05): Fix wrong / permissions
Closes #1098