armbian/build #1102: Orangepi-r1 crypt root error

xMase, 2018-09-08:
Hi when i try to compile armbian for orangepi-r1 with sd encryption an error come out. 
How can i resolve it?
`./compile.sh docker BOARD=orangepi-r1 BRANCH=next KERNEL_ONLY=no KERNEL_CONFIGURE=no RELEASE=stretch BUILD_DESKTOP=no CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="MYSECRECTPASS" CRYPTROOT_SSH_UNLOCK=yes CRYPTROOT_SSH_PORT=2222
`
> [ o.k. ] Creating partitions [ /boot: ext4 root: ext4 ]
> [ .... ] Encrypting root partition with LUKS... [ cryptsetup luksFormat /dev/loop0p2 ]
> Device /dev/loop0p2 doesn't exist or access denied.
> Device /dev/loop0p2 doesn't exist or access denied.
> [ o.k. ] Root partition encryption complete. 
> [ error ] ERROR in function check_loop_device [ image-helpers.sh:73 ]
> [ error ] Device node /dev/mapper/armbian-root does not exist 
> [ o.k. ] Process terminated 
> Cannot initialize device-mapper. Is dm_mod kernel module loaded?
> [ error ] ERROR in function unmount_on_exit [ image-helpers.sh:60 ]
> [ error ] debootstrap-ng was interrupted 
> [ o.k. ] Process terminated

igorpecovnik, 2018-09-08:
Probably you need to resolve kernel dependencies ...  like I did it here: https://github.com/armbian/build/commit/c504a1327801a7cb5bdad69de007b2c648cd5e71 ... please send PR when you get it working.

xMase, 2018-09-08:
i have already checked kernel dependency and it seems ok

igorpecovnik, 2018-09-08:
Then this is yet another Docker limitations which fine-tuning is not my strong point. It works when building native.

igorpecovnik, 2018-09-18:
It looks like we have a general problem with building image under Docker.

igorpecovnik, 2018-09-23:
Fixed https://github.com/armbian/build/commit/c1530db4820320a1e837901196d846b0ef71ee4c

xMase, 2018-09-25:
the problem persist:

> [ o.k. ] Creating partitions [ /boot: ext4 root: ext4 ]
> [ .... ] Creating device node [ /dev/loop0 ]
> [ .... ] Encrypting root partition with LUKS... [ cryptsetup luksFormat /dev/loop0p2 ]
> Device /dev/loop0p2 doesn't exist or access denied.
> Device /dev/loop0p2 doesn't exist or access denied.
> [ o.k. ] Root partition encryption complete. 
> [ error ] ERROR in function check_loop_device [ image-helpers.sh:73 ]
> [ error ] Device node /dev/mapper/armbian-root does not exist 
> [ o.k. ] Process terminated 
> Cannot initialize device-mapper. Is dm_mod kernel module loaded?
> [ error ] ERROR in function unmount_on_exit [ image-helpers.sh:60 ]
> [ error ] debootstrap-ng was interrupted 
> [ o.k. ] Process terminated 
>

igorpecovnik, 2018-09-25:
https://github.com/armbian/build/blob/master/lib/debootstrap-ng.sh#L425

I guess we need to wait for @zciendor for implementation.

zciendor, 2018-10-01:
> Device /dev/loop0p2 doesn't exist or access denied.

This seem to be the starting point of the problem. Do you have any possibility to narrow this error down in your build environment? `access denied` shouldn't be the case as the build script is run as root, right? root can access any loop device. At least when you run it natively the build script asks for sudo access at the beginning.

So the only possibility remaining is that `/dev/loop0p2` doesn't exist in your environment, which means there must have been another error before...

The two partitions are created here, which is agnostic to if encryption is used or not:
https://github.com/armbian/build/blob/575af9d4111058823d4d9f13fef4ad6b7fbabcdf/lib/debootstrap-ng.sh#L392

And probing for the partitions is done here, which is also agnostic to encryption:
https://github.com/armbian/build/blob/575af9d4111058823d4d9f13fef4ad6b7fbabcdf/lib/debootstrap-ng.sh#L413

zciendor, 2018-10-01:
> # TODO: pass /dev/mapper to Docker
I'm not very familiar with Docker... does you comment imply that `/dev/mapper` is not available inside a Docker container?

zciendor, 2018-10-01:
I'm not even sure if `cryptsetup` and `LUKS` can work from within a Docker container at all:
https://stackoverflow.com/questions/34439024/docker-with-device-mapper-and-luks

What I can definitely implement is better error handling or even a check that `CRYPTROOT_ENABLE=yes` is incompatible with the docker build environment. But that doesn't add a solution.

The alternative would be, that the partitioning, loop device and cryptsetup commands a not called from within the docker environment but before or outside of the Docker container, which even if it was possible would be a major rewrite of the build script...

igorpecovnik, 2018-10-01:
> The alternative would be, that the partitioning, loop device and cryptsetup commands a not called from within the docker environment but before or outside of the Docker container, which even if it was possible would be a major rewrite of the build script...

This is out of question.

> What I can definitely implement is better error handling or even a check that CRYPTROOT_ENABLE=yes is incompatible with the docker build environment. But that doesn't add a solution.

I know. If we don't have a solution, we can at least save people from frustration. I already added to the manual that CRYPTROOT_ENABLE is not supported under Docker.

closed 2020-01-07 by igorpecovnik