armbian/build #1103: need to set bloader parameter to boot emmc on Le Potato

PhracturedBlue, 2018-09-09:
Running Armbian 5.54 on Le Potato.

After running nand-sata-install, EMMC won't boot.  see:
https://forum.armbian.com/topic/5668-le-potato-writing-armbian-to-emmc/?do=findComment&comment=61454

The issue is that in u-boot 2018.5, the 'bloader' variable is coded to 
```bloader = ext4load mmc 0:1```
When booting from emmc, this should be:
```bloader = ext4load mmc 1:1```
I am not sure what the right place to solve this is, but my work around was to reset the variable in emmc:/boot/armbianEnv.txt after running ```nand-sata-install```

Tonymac32, 2018-09-10:
If you move to Armbian 5.59 the bootscript has been updated:

Note I am also assuming you're running Armbian official Le Potato image, not the tv box image 
https://github.com/armbian/build/blob/master/config/bootscripts/boot-meson64.cmd

PhracturedBlue, 2018-09-10:
I'm looking at that boot script and it looks the same as the one I have.  I don't see where bloader is being updated.  I can't actually test it since I already have the EMMC working

Tonymac32, 2018-09-10:
Hmmm, you're right, I had thought this had been updated, maybe it got lost in the "development" branch when the divergence became too much...

To your question, the armbianEnv.txt file is the appropriate place to change that from the user side.  Now that the eMMC apparently functions with our images (was not the case a while ago), we'll have to look into getting it properly supported.

lanefu, 2019-03-14:
is this issue still valid?

Tonymac32, 2019-03-15:
https://github.com/armbian/build/commit/cfd8a8956814a2f9da465769c1f9f5ad49e369cc

This should have been the fix.  I'd forgotten this issue, I was probably waiting on a tested-by from the op if possible.  Let me test it out again to make doubly sure.

lanefu, 2019-05-30:
closing old issue

closed 2019-05-30 by lanefu