armbian/build #1110: (regression?) latest version of nand-sata-install not listing NVMe as a target if it has no partitions

RossBencina, 2018-09-12:
Background: I found a problem while using the latest `master` branch version of  `nand-sata-install` to install Armbian onto an PCIe NVMe attached to a NanoPC-T4.

**Version Info**

The version of  `nand-sata-install` used was the latest from the `master` branch, containing recent changes by @ThomasKaiser to support NVMe install: specifically, this "consolidation" commit:

https://github.com/armbian/build/commit/7521bb7ff1e115f70fbc53990531276d8c018c71

I "installed" `nand-sata-install` by cloning the git repo, and copying `nand-sata-install` into `/usr/sbin`. I was careful to check that all permissions were preserved versus the system version.

The remainder of Armbian was an SD card containing Armbian_5.59_Nanopct4_Ubuntu_bionic_default_4.4.152_desktop

**Expected Behavior**

Run `nand-sata-install` and select NVMe as a target to install Armbian.

**Actual Behavior**

After booting from an sd card, with the NVMe device connected, `/dev` listed the following NVMe devices: `nvme0` and `nvme0n1`.  In other words, the NVMe device contained no partitions.

I then ran the `nand-sata-install` version referenced above.

I selected the option to boot from eMMC with system on NVMe.

Problem: the "Select destination" dialog box only gave me the option to install on `sdb1`. **It did not list any NVMe devices as a destination, nor did it give the option to partition the NVMe.**

**Workaround**

I was able to get `nand-sata-install` to recognise my NVMe by first running `gdisk` and creating a partition. Then when I ran the above version of `nand-sata-install` it worked.

**Analysis**

I believe that the problem is with the logic in `checkpartitions()`  (line 472 in the above commit)

```
	AvailablePartitions=$(lsblk -l | awk -F" " '/ part / {print $1}' | egrep '^sd|^nvme')
	if [[ -z $AvailablePartitions ]]; then
```

The problem is that if the sd card has a partition, then `AvailablePartitions` will be non-empty and so the user is not prompted to partition the NVMe. Then since the NVMe has zero partitions, the user is not prompted to select it as a target.

I think this was not a problem with earlier versions, because the user would select SATA/NVMe/etc from the main menu, so then `checkpartitions()` would only be checking if a particular type of device contained partitions.

igorpecovnik, 2018-09-12:
Working as expected. I made an image from sources and install it on the NVME drive. I did delete existing partitions. 
```
root@nanopct4:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1946376       0   1946376   0% /dev
tmpfs             390276    5384    384892   2% /run
/dev/nvme0n1p1 491207900 1047468 465138720   1% /
tmpfs            1951368       0   1951368   0% /dev/shm
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1951368       0   1951368   0% /sys/fs/cgroup
tmpfs            1951368       4   1951364   1% /tmp
/dev/mmcblk1p1  14796008   82860  13941828   1% /media/mmcboot
/dev/zram0         49584    2864     43136   7% /var/log
tmpfs             390272       0    390272   0% /run/user/0
```

ThomasKaiser, 2018-09-12:
> the "Select destination" dialog box only gave me the option to install on sdb1

`sdb1` is not SD card but on a NanoPC-T4 an USB device. So given that I tried to not alter any of the existing logic and just touched recognition of disks and partitions (replacing parsing `/proc/partitions` with `lsblk` calls) nothing changed. If there is an USB drive with a partition on it this one will be selected automagically.

BTW: I tried to do the same this morning (boot from eMMC with system on NVMe) and successfully bricked my NanoPC-T4 (no idea why, too lazy to connect serial console... but looking at leds and board behavior it's stuck in a reboot loop).

Two NanoPi M4 just arrived (with socketed eMMC) so maybe I get the culprit what went wrong. But looking at @igorpecovnik's output it seems he got it working from eMMC. So maybe I messed my installation up too much (fiddling around endlessly in `/boot/armbianEnv.txt`)

RossBencina, 2018-09-13:
> sdb1 is not SD card but on a NanoPC-T4 an USB device

I see. My mistake. I did also have a USB thumb drive plugged in. My "Analysis" is incorrect.

So the "bug" requires two preconditions:

- no partitions on the NVMe
- a partitioned USB thumb drive to be connected

I guess this is not really a bug, although the old UI where you select the type of target device first, and then check whether it was partitioned would not have exhibited this behavior.

Thanks for your rapid response.

RossBencina, 2018-09-13:
I added a second workaround

> Before running `nand-sata-install`, ensure that there are no USB storage devices connected to the board.

Perhaps "Select Destination" UI could be updated with a note: "NOTE: if your unpartitioned storage device does not show up in this list, please ensure that there are no extra partitioned storage devices connected."

Note also that I guess there would be a problem with the existing logic if the user tried to install to an unpartitioned SATA or USB device if there was a connected and partitioned NVMe.

ThomasKaiser, 2018-09-13:
> Note also that I guess there would be a problem with the existing logic if the user tried to install to an unpartitioned SATA or USB device if there was a connected and partitioned NVMe.

Sure. But that's a feature request / enhancement. When dealing with the `nand-sata-install` mess I always try to do minimal invasive changes not changing other parts of the logic for the simple fear of breaking stuff (the code is old and we should drop it and start from scratch as already several times discussed).

In general I agree. Would need a new issue with a feature request and ideally a PR with the appropriate changes. The logic `nand-sata-install` implements still cares only about old A20 boards with one NAND and one SATA device. Things have improved in the meantime so this needs adoption.

closed 2018-09-13 by ThomasKaiser