armbian/build #1121: Documentation - update requirements for "building armbian" needed

sfx2000, 2018-10-02:
Priority - Low
Severity - Minor

Please review/update the documentation website with new requirements.

https://docs.armbian.com/Developer-Guide_Build-Preparation/

For on-metal builds, docs.armbian is still saying Ubuntu 16.04LTS is the only supported release...

The tip still builds on 16.04LTS for now, but as the team has moved forward to 18.04LTS it should be consistent.

chwe17, 2018-10-04:
There you go:
https://github.com/armbian/documentation/blob/465e1333a5311b34cf14b6c27f6c9ad09128f7f4/docs/Developer-Guide_Build-Preparation.md
Docs can be forked and enhanced/corrected by everyone. :)

chwe17, 2018-10-07:
addressed here: https://github.com/armbian/documentation/pull/22/commits/0a6162b1898c14bc3ae7872c6857a51ea527b617

closed 2018-10-07 by chwe17