armbian/build #1125: Odroid C2: next and dev kernels do recognize USB hard-drives

getreu, 2018-10-04:
Odroid C2: next and dev kernels do recognize USB hard-drives

The stable kernel does.

chwe17, 2018-10-04:
from the text you erased when open an issue:

> Please note that this is a **build script** issue tracker.
> Issues with provided images should go to the Armbian forum.

I think this fits better in:
https://forum.armbian.com/forum/16-amlogic-s905x/
and: https://forum.armbian.com/topic/8309-usb-and-network-issues-with-mainline-kernel/ catches probably your issue.

closed 2018-10-04 by chwe17

getreu, 2018-10-05:
Done, I posted this issue here:
[Odroid C2: next and dev kernels do recognize USB hard-drives - Amlogic S905(x) - Armbian forum](https://forum.armbian.com/topic/8396-odroid-c2-next-and-dev-kernels-do-recognize-usb-hard-drives/)