armbian/build #1133: Tinker Board needs CONFIG_BT_HCIUART_3WIRE=y for bluetooth

EriHoss, 2018-10-14:
Hello, currently I am unable to get bluetooth working on Tinker Board unless I manually set the following kernel config options:

```
# Deps of 3WIRE
CONFIG_SERIAL_DEV_BUS=y
CONFIG_BT_HCIUART_SERDEV=y

# H5 Protocol Support
CONFIG_BT_HCIUART_3WIRE=y

# Asks for config (new stuff)
CONFIG_BT_HCIUART_NOKIA=n
CONFIG_BT_HCIUART_LL=n
CONFIG_BT_HCIUART_BCM=n
CONFIG_QCA7000_UART=n
CONFIG_SERIAL_DEV_CTRL_TTYPORT=n

```

Bluetooth works on TinkerBoard for me with this config, and fails without.

`/usr/bin/rtk_hciattach -n -s 115200 /dev/ttyS0 rtk_h5`

 fails because rtk_h5 (CONFIG_BT_HCIUART_3WIRE) is not supported by the kernel.

I am testing with Armbian 5.64, Kernel 4.14.76 and 4.18.14 on a Tinker Board (S)

Should this config be default for this device?

Tonymac32, 2018-10-14:
I didn't see, it got disabled for some reason here:
https://github.com/armbian/build/commit/980230cce1137db415ee32abc5c2745bfa9c95db#diff-822602262efe5145cb6d4cfd3d118dce

I had Bluetooth working on all flavors out of the box at one point, I'll have to go back through.  @igorpecovnik  put it back I see.

EriHoss, 2018-10-14:
Built 4.14.76 at 1c4340b. hcitool is finding hci0 now and scanning works. 

Tested out a bluetooth gamepad successfully.  

Adding these back appears to have done the trick on Tinker at least.

Tonymac32, 2018-10-14:
Good, it is the protocol for the BT uart, are you triggering the hcitool
command line manually?  There should be a rule duing at boot.  I'll have to
check.

igorpecovnik, 2018-10-14:
CLI images lack BT packages. Perhaps that is the case?

Tonymac32, 2018-10-14:
Hmmm, the specific rtk_hciattach is part of the bsp, so it will (should) be
in all images

EriHoss, 2018-10-14:
there is `/usr/bin/rtk_hciattach` that exists in the image already. It is called by `tinker-bluetooth.service` on boot.

However bluez is not installed on CLI like mentioned. I installed it using `apt install bluetooth` which brought it in.

Should give you hcitool.

I am just verifying by running hcitool.

`hcitool devices` to list the controller (hci0) and `hcitool scan` to list nearby bluetooth devices.

igorpecovnik, 2018-11-08:
Fixed with https://github.com/armbian/build/commit/2ceaa6e8bd89abf576b65a0a60b8356b8b5f74df

closed 2018-11-08 by igorpecovnik